yaqwsx/KiKit #1: Reconstruct arcs in Edge.Cuts

yaqwsx, 2020-04-07:
Current implementation removes arcs and circles in the internal representation of a substrate and approximates them with lines (mimicking what Shapely does). It would be nice to reconstruct arcs when exporting the board.

The challenges are:
- we can test if consequential points lie on the same circle, however, then we may deduce false arcs - e.g. if the input board was a regular polygon
- we can store a list of original arcs and match them back - however, parts of the arcs might be lost during substrate merging

yaqwsx, 2020-04-10:
The first draft implemented in [circleReconstruction](/yaqwsx/KiKit/tree/circleReconstruction). The implementation works, but there are some unresolved corner cases that make things fail terribly.

lanserge, 2020-10-25:
One of the possibilities is to use FreeCad for contours. This will do everything in pure arc segments. Here some of my code that could be helpful:

import sys
sys.path.append('/usr/lib/freecad-python{}/lib'.format(sys.version_info[0]))
import Part, Base

def create_cuts(pts,r): # pts define path, r - the radius for contour around path
    pts=list(map(lambda pt:Base.Vector(pt[0],pt[1],0),pts)) # convert xy pairs to Vector in 3d space with z=0
    if len(pts)==2: # will not work with one segment, adding segment back to start
        pts.append(pts[0])
    pairs=list(zip(pts[:-1],pts[1:])) # create points pairs for each segment
    obj=Part.Wire(list(map(lambda pair:Part.makeLine(*pair),pairs))) # create wire of line segments connecting each two points
    contour=obj.makeOffset2D(r,join=0,fill=False,openResult=False,intersection=True)#create contour with radius r around path
    for e in contour.Edges: # now go through the edges of the contour
        pt1=e.firstVertex().Point # segment start
        pt2=e.lastVertex().Point # segment end
        if isinstance(e.Curve,Part.Line):
            # line segment
            print("line from {} to {}".format(tuple(pt1),tuple(pt2)))
        elif isinstance(e.Curve,Part.Circle):
            # arc segment
            ptc=e.Curve.Center
            r=e.Curve.Radius
            print("arc from {} to {} with center at {} and radius {}"
                  .format(tuple(pt1),tuple(pt2),tuple(ptc),r))

lanserge, 2020-10-25:
I have added code for merging contours. Bellow is working code with test example function.

```
def create_cuts(pts,r): # pts define path, r - the radius for contour around path
    pts=list(map(lambda pt:Base.Vector(pt[0],pt[1],0),pts)) # convert xy pairs to Vector in 3d space with z=0
    if len(pts)==2: # will not work with one segment, adding segment back to start
        pts.append(pts[0])
    pairs=list(zip(pts[:-1],pts[1:])) # create points pairs for each segment
    wire=Part.Wire(list(map(lambda pair:Part.makeLine(*pair),pairs))) # create wire of line segments connecting each two points
    contour=wire.makeOffset2D(r,join=0,fill=False,openResult=False,intersection=True)#create contour with radius r around path
    return contour

def merge_contours(cont1,cont2):
    cont1=Part.Face(cont1)
    cont2=Part.Face(cont2)
    join_cont=cont1.fuse(cont2).removeSplitter()
    if len(join_cont.Faces)!=1:
        raise Exception("Contours do not intersect")
    return join_cont.Faces[0].OuterWire

def print_contour(contour):
    for e in contour.Edges: # now go through the edges of the contour
        pt1=e.firstVertex().Point # segment start
        pt2=e.lastVertex().Point # segment end
        if isinstance(e.Curve,Part.Line):
            # line segment
            print("line from {} to {}".format(tuple(pt1),tuple(pt2)))
        elif isinstance(e.Curve,Part.Circle):
            # arc segment
            ptc=e.Curve.Center
            r=e.Curve.Radius
            print("arc from {} to {} with center at {} and radius {}"
                  .format(tuple(pt1),tuple(pt2),tuple(ptc),r))

def test_2d():
    r=5
    cut1=create_cuts([(0,0),(10,0)],r)
    cut2=create_cuts([(10,0),(10,10)],r)
    join_cut=merge_contours(cut1,cut2)
    print_contour(join_cut)
```

yaqwsx, 2020-10-26:
Thank you for your proposal, however, I will not use it in KiKit for several reasons:

- FreeCAD is a heavy dependency, which cannot be automatically installed via PIP. It leads to poor user experience and it is another potential source of problems like #21. We would need to require that KiCAD and FreeCAD use the same minor version of Python which might be troublesome.
- When the BRep representation via OpenCascade is used (this is what FreeCAD internally does) it makes sense to rewrite the whole panelization feature of KiKit to use it instead of Shapely. It is a lot of work I cannot schedule right now and I am not sure if it is worth it.

I hope you understand.

lanserge, 2020-10-26:
I understand. I am in the same shoes.
The FreeCad api is very close to OpenCascade and all of them are open source, hope my code above could be helpful when dealing with OpenCascade implementation (it is possible to dig from my code down to OpenCascade low level through open source code).

yaqwsx, 2020-10-26:
I know OpenCascade, however, it has exactly the same problems as FreeCAD. You cannot install it via pip since it brings non-trivial binary dependency. I know only about distribution via Anaconda, which not only I dislike, but it will also prevent integration with KiCAD as it creates custom venv and brings its version of Python. Also, I would like to make my life easier with Windows support - it is already pretty complicated because of Shapely. OpenCascade would make the situation even worse.

sslupsky, 2021-02-27:
BREP is used widely in the commercial 3D modelling space as well.  BREP is also forms the foundational data format for STEP files which are an ISO standard.

joaopedrovbs, 2022-04-01:
Any updates on this? I used the Panelize feature to create a panel and the STEP File export crashed because the panel has a lot of arcs that were converted into lines.

yaqwsx, 2022-04-02:
There's no progress at the moment on this feature as it wasn't high on the priority - there are no know cases where the manufacturer would complain or delivered defective/poor quality boards. The current approach has only a few drawbacks - it bloats the filesize and the STEP export doesn't contain arcs, thus it is harder to use when designing, e.g., an enclosure.

I am not saying this feature will not land in KiKit - actually, I have several ideas for implementing it much better than the original proposal in https://github.com/yaqwsx/KiKit/tree/circleReconstruction; at the moment it just has a very low priority.

PS: If the STEP exporter is crashing, you should file a bug report for KiCAD.

joaopedrovbs, 2022-04-04:
> PS: If the STEP exporter is crashing, you should file a bug report for KiCAD.

Thanks for the info @yaqwsx, will file a bug report right away.  Also, thanks for the work on KiKit, it's awesome!

yaqwsx, 2022-04-27:
The new implementation is available in #347. @joaopedrovbs please, give it a try and report any problems.

closed 2022-05-08 by yaqwsx