yaqwsx/KiKit #7: Problems with CLI

colesnicov, 2020-04-09:
Hello.  I type this:

`$ kikit panelize grid --gridsize 2 2 ./atmega328pb.kicad_pcb ./grb/test.grbl`

I see this:
```
Traceback (most recent call last):
  File "/home/denis/.local/bin/kikit", line 11, in <module>
    sys.exit(cli())
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/denis/.local/lib/python3.6/site-packages/kikit/ui.py", line 69, in grid
    outerHorTabThickness=oht, outerVerTabThickness=ovt)
  File "/home/denis/.local/lib/python3.6/site-packages/kikit/panelize.py", line 551, in makeGrid
    sourceArea, tolerance, verSpace, horSpace)
  File "/home/denis/.local/lib/python3.6/site-packages/kikit/panelize.py", line 414, in _placeBoardsInGrid
    dest = self._boardGridPos(destination, i, j, boardSize, horSpace, verSpace)
  File "/home/denis/.local/lib/python3.6/site-packages/kikit/panelize.py", line 402, in _boardGridPos                                                            
    return wxPoint(destination[0] + j * (boardSize.GetWidth() + horSpace - FromMM(0.001)),                                                                       
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType' 
```
If I type this:

`$ kikit panelize grid --gridsize <2 2> ./atmega328pb.kicad_pcb ./grb/test.grbl`
I get this:

`bash: 2: Adresář nebo soubor neexistuje`

So, what nextz..?

lukasz-skalski, 2020-04-09:
@colesnicov first of all, add --space argument, for example:

kikit panelize grid --space 0 --gridsize 2 2 ./atmega328pb.kicad_pcb ./grb/test.kicad_pcb

What's more, output file should be also a *.kicad_pcb file - not *.grbl as in your case.

colesnicov, 2020-04-09:
Thank you, it's perfect

`kikit panelize grid --space 2 --gridsize 2 2 --tabwidth 5.0 --tabheight 5.0 --mousebites 0.4 1.6 --radius 0.9 ./atmega328pb.kicad_pcb ./grb/test.kicad_pcb` - OK.

But if I'll put panelsize options argument like this:
`kikit panelize grid --space 2 --gridsize 2 2 --tabwidth 5.0 --tabheight 5.0 --mousebites 0.4 1.6 --radius 0.9 --panelsize 150.0 110.0 ./atmega328pb.kicad_pcb ./grb/test.kicad_pcb`

It is looks that like a mismatch 
```
Traceback (most recent call last):
  File "/home/denis/.local/bin/kikit", line 11, in <module>
    sys.exit(cli())
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/denis/.local/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/denis/.local/lib/python3.6/site-packages/kikit/ui.py", line 59, in grid
    w, h = panelize
TypeError: 'Group' object is not iterable
```

lukasz-skalski, 2020-04-09:
There is a typo in variable name (panelize->panelisize). I've already send pull requests for it:

https://github.com/yaqwsx/KiKit/pull/6

I hope that author soon will merge it.

colesnicov, 2020-04-09:
Thank you for the early reaction. I edited the local file and it works.

closed 2020-04-09 by colesnicov