yaqwsx/KiKit #10: Problem panelizing circular board

EnJens, 2020-04-16:
When trying to panelize the board available in this github repository: https://github.com/EnJens/ilulissat (a fully circular board), 

I'm using a trivial script based on the example for initial testing:
```python
from kikit.panelize import Panel, fromMm
from pcbnew import *
panel = Panel()
size, cuts = panel.makeGrid("ilulissat.kicad_pcb", 4, 3, wxPointMM(100, 40),
            tolerance=fromMm(5), verSpace=fromMm(5), horSpace=fromMm(5),
            outerHorTabThickness=fromMm(3), outerVerTabThickness=fromMm(3),
            verTabWidth=fromMm(15), horTabWidth=fromMm(8),
            radius=fromMm(1))
panel.makeMouseBites(cuts, diameter=fromMm(0.5), spacing=fromMm(1))
panel.makeFrame(size, fromMm(100), fromMm(100), fromMm(3), radius=fromMm(1))
panel.save("panel.kicad_pcb")
```

Running the script I get the error:
```
Traceback (most recent call last):
  File "panelize.py", line 4, in <module>
    size, cuts = panel.makeGrid("ilulissat.kicad_pcb", 4, 3, wxPointMM(100, 40),
  File "kikit/panelize.py", line 553, in makeGrid
    cuts = self._makeSingleInnerTabs(gridDest, rows, cols, boardSize,
  File "kikit/panelize.py", line 473, in _makeSingleInnerTabs
    self.appendSubstrate(tab, radius)
  File "kikit/panelize.py", line 313, in appendSubstrate
    for geom in filletCandidates:
TypeError: 'LineString' object is not iterable
```

yaqwsx, 2020-04-17:
Thank you for reporting and for giving me a proper example to examine the problem. 

The error you see is caused by the fillet function (this is a bug and I have to fix it). If you remove radius, you get a panel: [panel.zip](https://github.com/yaqwsx/KiKit/files/4491090/panel.zip)

```
from kikit.panelize import Panel, fromMm
from pcbnew import *
panel = Panel()
size, cuts = panel.makeGrid("ilulissat.kicad_pcb", 3, 3, wxPointMM(100, 40),
            tolerance=fromMm(5), verSpace=fromMm(5), horSpace=fromMm(5),
            outerHorTabThickness=fromMm(3), outerVerTabThickness=fromMm(3),
            verTabWidth=fromMm(15), horTabWidth=fromMm(8))
panel.makeMouseBites(cuts, diameter=fromMm(0.5), spacing=fromMm(1))
panel.makeFrame(size, fromMm(100), fromMm(100), fromMm(3))
panel.save("panel.kicad_pcb")
```

However, `makeGrid` assumes there is a straight edge to which attach the tab - which there isn't. In the panel produced above, you only get a point contact. This is why I on twitter mentioned you need to write a custom script and not use the CLI interface - you simply have to manually add pieces of the substrate which are wider and reach the interior of the board. 

This brings me to a question - what is the expected result of paneling circular boards? Is it tabs that reach to the boar outline and mouse bites copying the original outline? Is it something different?

EnJens, 2020-04-17:
ah, i didn't realize it had to be that manual script :)
I understand it's not a trivial problem though.

What I would expect is exactly what you are suggesting yes - tabs that go to the outline and mouse bytes roughly following the original outline. For small tabs, the local gradient should be sufficiently small it won't be a huge issue. I think this is how oshpark does it as well. 

Tbh, as a start, i think for 99% of things (assuming small tabs), it would even work if the tabs just extended to the board outline and then the mouse bytes on a straight line.

yaqwsx, 2020-04-18:
I am nearly finished with the reimplementation of the tab feature. Now, it should work with an arbitrary board shape. It also allows you to place multiple tabs and approximate circles with V-Cuts. It also solves #2 and #5. The results for your board are below. Before I push the changes, I have to adapt CLI for the new features (I will push the changes later today).

[image]
[image]

yaqwsx, 2020-04-18:
The new implementation is pushed, but not yet released. You should be able to create the panels from CLI. Could you test it? I would appreciate tests even on other boards. Once some else besides me confirm, it works, I will release it.

To install the development version call:
```
pip3 install --force --no-deps git+https://github.com/yaqwsx/KiKit@master
```

EnJens, 2020-04-18:
I'll try and see if I can give it a try later today.

I noticed one minor thing from the pictures - it looks like the drill hits are a bit far in? As in on the inside of edge cut rather than outside which might cause trouble with parts close to the edge (like the chip antenna on the bottom of the board).

yaqwsx, 2020-04-18:
It is on purpose - most of the manufacturers state in their guidelines that the mouse bites should be moved 0.25 mm inside the board. This parameter is configurable (`--mousebites <drill_size> <spacing> <offset>`).

EnJens, 2020-04-18:
Hi,
I've played around a bit with a few different boards.
Ilulissat:
[image]

Random APA102 mini-board:
[image]

I also tried it with an old weird board (https://github.com/EnJens/hyperstm) where it failed quite a lot:
```
$ python3 -m kikit.ui panelize grid -g 1 1 -p 100.0 100.0 -s 2.0 --tabheight 2.5 --tabwidth 2.5 --mousebites 0.5 1.0 -0.25 hyperstm.kicad_pcb panel.kicad_pcb

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "kikit/ui.py", line 186, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "kikit/ui.py", line 73, in grid
    psize, cuts = panel.makeGrid(input, rows, cols, wxPointMM(50, 50),
  File "kikit/panelize.py", line 578, in makeGrid
    boardSize = self._placeBoardsInGrid(boardfile, rows, cols, destination,
  File "kikit/panelize.py", line 417, in _placeBoardsInGrid
    boardSize = self.appendBoard(boardfile, dest, sourceArea=sourceArea,
  File "kikit/panelize.py", line 309, in appendBoard
    self.boardSubstrate.union(Substrate(edges, bufferOutline))
  File "kikit/substrate.py", line 321, in __init__
    polygons = [toShapely(ring, geometryList) for ring in extractRings(geometryList)]
  File "kikit/substrate.py", line 74, in extractRings
    raise RuntimeError("Wrong number of entities ({}) at [{}, {}]".format(
RuntimeError: Wrong number of entities (1) at [120.8796, 94.1391]
```
It's a bit of a weird board though so maybe it's not really realistic :)
I also get the same issue when trying someone elses board: https://bitbucket.org/raytrails/raybeacon.git but that isn't really super well suited for panelizing either.

Overall, I'd say it works really well!

yaqwsx, 2020-04-19:
The errors you see mean that there were slight discontinuities in your outline. I made the errors easier to read in c1a503a9d2755a4c466802d23d2833e234b11e2c Now you get errors like:
```
python3 -m kikit.ui panelize grid -g 1 1 -p 100.0 100.0 -s 2.0 --tabheight 2.5 --tabwidth 2.5 --mousebites 0.5 1.0 -0.25 /tmp/hyperstm/hyperstm.kicad_pcb panel.kicad_pcb
An error occurred: /tmp/hyperstm/hyperstm.kicad_pcb: Discontinuous outline at [185.599982, 91.850023]
No output files produced
```
The problem is with two consequential arcs in KiCAD - it is practically impossible to arrange them such that the starting and ending points overlap.

[image]

3D viewer probably has large tolerance - that's why it works there. However, STEP export does not work for your board. I decided not to implement a large threshold as some weird behavior might pop-up in the future due to this. As I was advised on KiCAD IRC channel - always put a short line segment between two arcs.

When I fixed the errors, it works:

[image]
[image]

Therefore, I consider this issue resolved. If you find more boards where it does not work, please let me know by opening a new issue.

closed 2020-04-19 by yaqwsx