yaqwsx/KiKit #20: Mousebites issues?

rcludwick, 2020-04-27:
You don't specify a default here, so maybe this is just a documentation issue.  

I've made boards where I have traces right up against the edge of the board.  A positive offset on the mousebites might be problematic as that could cut into traces.  A positive mousebite offset would cut into those traces.  

So in your example ```--mousebites 0.5 1 0.25``` produces this.
[image]

If offset is negative, the mousebites move off the board and back into the tab.  

I was thinking the default offset for mousebites would be 1/2 the radius of the mousebites so the edge of the mousebites would be the normal edge of the board.

yaqwsx, 2020-04-27:
You are right; there is no mouse bite default setting. What I state in the example is the recommended value by several manufacturers.

Feel free to open PR for the default setting, but please be sure that the PR includes proper documentation on the defaults.

rcludwick, 2020-04-27:
But does that offset go inside the board edge or outside of it?  That's the question I have.  Right now they're on the inside.  It seems like mousebites could go anywhere, right?  you could put them in the middle of the tab if you wanted... 

Granted it's a preference, but I prefer the one with the least surprise to the end user.  I was surprised by it for sure.

yaqwsx, 2020-04-27:
The guidelines say it goes *inside* the board. The reason for that is that once you break the mouse bites, you want as least material outside the designed board outline and there is less material to clean up. 

See: https://www.allpcb.com/mouse_bites_pcb.html and https://blogs.mentor.com/tom-hausherr/blog/tag/mouse-bite/

yaqwsx, 2020-04-27:
Also, maybe we don't want a default value for mouse bites - in this ways, the user is not surprised (and novice will follow the recommended values)

rcludwick, 2020-04-27:
I think then it's best add some good documentation about this behavior.   I would certainly appreciate it.   Offset behavior seems like a subtle thing but could have a big impact on the design of a board, especially when designing around mousebites then.

WebDust21, 2020-06-25:
> You don't specify a default here, so maybe this is just a documentation issue.
> 
> I've made boards where I have traces right up against the edge of the board. A positive offset on the mousebites might be problematic as that could cut into traces.
> 
> So in your example `--mousebites 0.5 1 0.25` produces this.
> [image]
> 
> If offset is negative, the mousebites move off the board and back into the tab.
> 
> I was thinking the default offset for mousebites would be 1/2 the radius of the mousebites so the edge of the mousebites would be the normal edge of the board.

Thanks for the help for a newbie to KiKit...that was exactly the bit of info I needed.  Trying to panelize a board, and the example panelization setting (positive offset) obliterates traces on the edges of the PCB.  Flipping the offset to negative...bingo, problem solved.

yaqwsx, 2020-06-27:
I added a section about this into [FAQ](doc/faq.md) in 39c0dfc. Therefore, I consider this issue as resolved.

closed 2020-06-27 by yaqwsx