yaqwsx/KiKit #21: ModuleNotFoundError: No module named 'pcbnew'

lazyzero, 2020-04-27:
Hello,

I tried kikit on Ubuntu 18.04 and MacOs. I installed it one time from the repository with python3 install and one time from pip3. In both cases and with several kicad_pcb files I get the following error message. Is there a pcbnew package I have install?

Best Christian

```
/usr/local/bin/kikit panelize grid --space 2 --gridsize 2 2 --tabwidth 3 --tabheight 3 --mousebites 0.5 1 0.25 --radius 1 --panelsize 80 80 /tmp/test.kicad_pcb /tmp/panel.kicad_pcb
Traceback (most recent call last):
  File "/usr/local/bin/kikit", line 11, in <module>
    load_entry_point('KiKit==0.3', 'console_scripts', 'kikit')()
  File "/home/user/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/user/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/home/user/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/home/user/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.6/dist-packages/KiKit-0.3-py3.6.egg/kikit/ui.py", line 2, in <module>
    import kikit.export as kiexport
  File "/usr/local/lib/python3.6/dist-packages/KiKit-0.3-py3.6.egg/kikit/export.py", line 5, in <module>
    from pcbnew import *
ModuleNotFoundError: No module named 'pcbnew'
```

yaqwsx, 2020-04-27:
What version of KiCAD do you have on the target systems?

lazyzero, 2020-04-27:
On Linux it is 4.0.7
On Mac I use 5.1.4

ameisso, 2020-04-27:
Hello, 
I'm having the exact same issue with Kicad 5.1.5 (latest) on mac. 
Best

yaqwsx, 2020-04-27:
@lazyzero: KiCAD 4 is not supported (it uses Python 2, not Python 3), please try version 5 on Linux.

There might be a problem with the Python version on Mac. Could you try to open an interactive python shell for both Python 2 and Python 3 (probably `python` and `python3` commands) and enter the following command and post its output?

```python
import pcbnew
```

Unfortunately, I don't have a Mac to try it and debug by myself

ameisso, 2020-04-27:
Hello, 

here is the result : 
```
>>> import pcbnew
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pcbnew'
```

for info, pcbnew is located here on my computer : 

> /Applications/KiCad/pcbnew.app/Contents/MacOS/pcbnew

Best

yaqwsx, 2020-04-27:
Is it the output of Python 2 or 3? I need output from *both*.

ameisso, 2020-04-27:
It's the same for both, 

here is a complete transcript : 

```
> Python 3.7.6 (default, Dec 30 2019, 19:38:26) 
> [Clang 11.0.0 (clang-1100.0.33.16)] on darwin
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import pcbnew
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ModuleNotFoundError: No module named 'pcbnew'
> >>> 
> KeyboardInterrupt
> >>> 
```

```
> 
> WARNING: Python 2.7 is not recommended. 
> This version is included in macOS for compatibility with legacy software. 
> Future versions of macOS will not include Python 2.7. 
> Instead, it is recommended that you transition to using 'python3' from within Terminal.
> 
> Python 2.7.16 (default, Feb 29 2020, 01:55:37) 
> [GCC 4.2.1 Compatible Apple LLVM 11.0.3 (clang-1103.0.29.20) (-macos10.15-objc- on darwin
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import pcbnew
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ImportError: No module named pcbnew
> 
```

lazyzero, 2020-04-27:
Same outputs for me.

Mean-while I overcome the problem by using the Dockerfile. I build it on my Linux machine and ran it with -i -t and copied my pcbnew files in the container.

yaqwsx, 2020-04-27:
I am afraid that KiCAD might be packed with its own Python binary, let me check with KiCAD developers.

As @lazyzero one possibility on MacOS would be to use the Docker image `yaqwsx/kikit` I provide. I use it in CI to build things. There is also no need to copy the files into a container - you can mount a directory to the container. See [travis.yml](https://github.com/RoboticsBrno/RB0002-BatteryPack/blob/master/.travis.yml)

Astroghost-Francois, 2020-04-28:
Hi , unable to use it on windows 10 , same error.
Regards
François

yaqwsx, 2020-04-28:
...and the same cause. Unfortunately, there's no quick fix since KiCAD scripting is quite messy as I have found out - on Windows, KiCAD brings its own interpreter of Python, which is not able to use non-purely-python dependencies.

I have some ideas to overcome this issue, but it won't be any time soon, unfortunately.

In the meantime, I can recommend using KiKit via Docker image `yaqwsx/kikit` - with the recent updates of Docker on Windows it should work. Also, you can install KiCAD and KiKit in WSL, where it should work out of the box.

yaqwsx, 2020-04-28:
@lazyzero Can I ask you to check the Python version used by KiCAD on MacOS? When you run "Tools->Scripting console", what is the output?

kortas87, 2020-05-03:
I solved this removing version 5.0.x and installing version 5.1.5 (https://kicad-pcb.org/download/ubuntu/) - system xubuntu 18.04

yaqwsx, 2020-06-27:
I added a note about the situation with Windows into [FAQ](doc/faq.md). Unfortunately, there is nothing I can do about it until KiCAD 6.0 is released.

Gasman2014, 2020-07-02:
It would be worth adding to the note to say the same applies to macOS. The macOS builds use a KiCad packaged version of Python 2.7 which is (normally) installed in /Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/python

ginkoblongata, 2020-07-07:
I have a fresh install of openSuse 15.2 and KiCad 5.1.5.

This issue is happening here too.

yaqwsx, 2020-07-07:
@ginkoblongata Please open Pcbnew, go to Help -> About and click Copy version info and paste it here.

ginkoblongata, 2020-07-07:
Application: Pcbnew
Version: 5.1.5-5.1.5, release build
Libraries:
    wxWidgets 3.0.3
    libcurl/7.66.0 OpenSSL/1.1.1d-fips zlib/1.2.11 libidn2/2.2.0 libpsl/0.20.1 (+libidn2/2.2.0) libssh/0.8.7/openssl/zlib nghttp2/1.40.0
Platform: Linux 5.3.18-lp152.19-default x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.3 (wchar_t,STL containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.66.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

ginkoblongata, 2020-07-07:
Also, my python3:
==================================
~/test> which kikit
/usr/bin/kikit
~/test> cat /usr/bin/kikit
#!/usr/bin/python3
# -*- coding: utf-8 -*-
import re
import sys
from kikit.ui import cli
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(cli())
~/test> python3 --version
Python 3.6.10
~/test>

yaqwsx, 2020-07-07:
Could you also go to the scripting console (Tools -> Scripting console) and send me the output of the following commands (you have to put them one by one into the interpreter):
```
import sys
sys.version
sys.executable
```

ginkoblongata, 2020-07-07:
`Py 0.9.8
Python 2.7.17 (default, Oct 24 2019, 12:43:41) [GCC] on linux2
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/levi/.config/kicad/PyShell_pcbnew_startup.py
import sys
sys.version
'2.7.17 (default, Oct 24 2019, 12:43:41) [GCC]'
sys.executable
'/usr/bin/python'
`

yaqwsx, 2020-07-07:
Unfortunately, KiCAD on OpenSuse is compiled against Python2, but you run KiKit inside Python3. You could try to install KiKit using pip for Python2, but I cannot guarantee you that it will work - and I do not plan to support Python2 for KiKit.

To use KiKit I see two options:
- use the docker image I provide
- compile KiCAD for your system with Python3 support
- ask package maintainers of your distribution if they could support KiCAD with Python3 support.

ginkoblongata, 2020-07-07:
In the scripting console I am able to do:

`>>> import pcbnew`

and it does not error.

I naively attempted to change the /usr/bin/kikit file to use python instead of python3, but it did not work.

```
# after changeing to python instead of python3 in kikit file:
~/test> kikit panelize
Traceback (most recent call last):
  File "/usr/bin/kikit", line 5, in <module>
    from kikit.ui import cli
ImportError: No module named kikit.ui
~/test> sudo vi /usr/bin/kikit

# after reverting kikit file to python3:
~/test> kikit panelize
Traceback (most recent call last):
  File "/usr/bin/kikit", line 5, in <module>
    from kikit.ui import cli
  File "/usr/lib/python3.6/site-packages/kikit/ui.py", line 2, in <module>
    import kikit.export as kiexport
  File "/usr/lib/python3.6/site-packages/kikit/export.py", line 5, in <module>
    from pcbnew import *
ModuleNotFoundError: No module named 'pcbnew'
~/test>
```

ginkoblongata, 2020-07-07:
That makes sense.

I think that the docker image probably has the best chance of success for my case given my inexperience with python2 vs 3.

I am curious tho, are you implying that kicad is using python 3 some other distros?
I got the impression that kikit is for kicad exclusively and that kikit always has been python3.

Is that correct?

yaqwsx, 2020-07-07:
KiCAD uses SWIG for the Python bindings. Due to Python's design, the SWIG bindings have to be compiled for an exact Python interpreter version (the Python interpreter can't load a library compiled for different versions - this includes the minor revisions). It is up to the person which compiles the KiCAD which version of Python it will use. Usually, it is the system defaults Python. I am surprised that a recent version of OpenSUSE uses Python 2. All the other Major distributions (Ubuntu, Fedora, Arch) already use Python3 with KiCAD for a couple of years.

ginkoblongata, 2020-07-07:
Thank you! Works great!

runger1101001, 2020-10-15:
+1 on this as a MacOS user... will work around it using the docker image for now.

yaqwsx, 2021-03-11:
Great news! KiCAD 6 will be available with MSVC build instead of MSYS2 build, therefore, KiKit will work on Windows out of the box (just the installation might be a little bit complicated - about 3-4 more commands compared to Linux). The same should go with MacOS

exislow, 2021-12-09:
I do have se same problem on macOS 11.6.1, KiCad 5.99 and python3.9. Is there any solution for this?

yaqwsx, 2021-12-09:
Does your PYTHONPATH environmental variable point to the directory with the pcbnew? The automatic detection of the environment has not been added.

exislow, 2021-12-09:
I did this afterwards. But now I get this error:

```
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 2): Library not loaded: @rpath/libwx_osx_cocoau_gl-3.1.5.dylib
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: image not found
No output files produced
```

Any ideas?

yaqwsx, 2021-12-09:
Unfortunately, I don't have Mac environment available nor I am a Mac user so it is hard for me to tell. Based on quick googling I am under the impression that you are trying to use KiKit with KiCAD v5. This is, unfortunately, not supported due to the problems you observe.

BTW: I also found the following; no idea if it is a solution (or a good idea to run it):
- https://gitlab.com/kicad/code/kicad/-/issues/3936
- https://forum.kicad.info/t/problem-running-pcbnew-python-plugin-on-os-x/6617
- https://gist.github.com/brainstorm/8d9b10d8a250354edf16fae1bc1a672e

runger1101001, 2021-12-09:
Hey @exislow , I tried for ages to get KiKit working on OS X, and it never worked, there was always some issue due to the dual pythons, the embedded python stuff, the OS X security, the paths, etc, etc...
In the end I just use the docker version of KiKit under OS X. This worked like a charm first try. Myadvice therefore: save yourself the pain and just use the docker version.

a7hybnj2, 2021-12-09:
@runger1101001 I had it running once upon a time but I have been running kicad nightlies for years and it eventually broke. I am just now trying to get it going again. Maybe if we work together we can figure it out. I just started trying with current nightly and I guess I will post my results here.

a7hybnj2, 2021-12-10:
Well, I just got to the same place @exislow is: `libwx_osx_cocoau_gl-3.1.5.dylib`

```
kikit panelize --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts conn.kicad_pcb panel.kicad.pcb
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/libwx_osx_cocoau_gl-3.1.5.dylib
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/usr/local/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file), '/usr/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file)
No output files produced
```

a7hybnj2, 2021-12-10:
Okay. I got one step further: `brew install wxwidgets` populated the `libwx_osx_cocoau_gl-3.1.5.dylib` file but now my error is
```
kikit panelize --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts conn.kicad_pcb panel.kicad.pcb
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/libwx_osx_cocoau-3.1.5.dylib
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file)
No output files produced
```
which looks the same but it is the base sans `gl`

exislow, 2021-12-10:
> Unfortunately, I don't have Mac environment available nor I am a Mac user so it is hard for me to tell. Based on quick googling I am under the impression that you are trying to use KiKit with KiCAD [...]

I appreciate your effort. Unfortunately none of the links had a proper solution. The last one, however, look promising but it does not fix the `*.dylib` errors.

runger1101001, 2021-12-10:
Personally, I will try again when KiCad 6 is officially out. There have been repeated changes to the python stuff, and an effort for a whole new Python API is underway (though it won't be ready for KiCad 6) and IMHO it simply isn't worth trying to fix things until they make the KiCad OS X python support stable.
And in the meantime, using docker is really quite a simple and good solution...

a7hybnj2, 2021-12-10:
Well, I am disappointed in not getting this running natively but I did install the docker:nightly which is working fine. I have done some searching for the file `libwx_osx_cocoau-3.1.5.dylib` and it doesn't appear to exist anywhere. I am wondering if it is a typo somewhere...

yaqwsx, 2021-12-18:
The problem should be resolved with #238 being merged and KiCAD 6 release. I am closing this (a quite old issue). If there will be any problems on macOS after v6 is released, please open a new issue.

closed 2021-12-18 by yaqwsx

runger1101001, 2021-12-19:
I'm looking forward to trying it out! Thanks for your support throughout!

a7hybnj2, 2021-12-20:
Hey great! It does seem like we made it past `this` issue but I did reference this in #252 as there are still some unusual things on macOS.