yaqwsx/KiKit #22: Wrong X/Y when extracting a board -> misleading error

benjaminaigner, 2020-04-28:
Hi,

THX for this nice tool!

When extracting a board and you don't give the correct corner for extracting, you will get following error:
`An error occurred: ../breakout.kicad_pcb: Discontinuous outline at [X,Y]`

Please clarify in the example that the upper left corner is necessary for extracting the board.

Greetings

yaqwsx, 2020-04-28:
The error says that within the source area the board outline is discontinued, therefore, we cannot reconstruct the PCB substrate and hence continue. 

Currently, I have no idea how to perform an algorithmic sanity check, that the user indeed specified the source area as he intended. If you have any ideas on how to perform such a check, they are welcomed.

About clarification - I am happy to improve the documentation. Could you just be more specific what change should we apply? Preferably, give me the exact points in documentation/program help which are confusing.

benjaminaigner, 2020-04-28:
https://github.com/yaqwsx/KiKit/blob/master/doc/examples.md#extract-board

"Select the X & Y coordinates from the upper left corner of the area you want to select."

I just thought it is not necessary to create a fork for just one line :-).

The problem with this error message: it was pointing to a corner on my PCB, so I fiddled around with the Edge-Cut lines because I thought they are not fully closed.

yaqwsx, 2020-04-28:
I have one more question for you, as a user. Currently, the source area behaves that it includes only board elements that can fully fit inside. Is this behavior OK for you?

I am happy to include the change you requested.

benjaminaigner, 2020-04-28:
This is another good question :-).
I think it would be good to include other parts as well, for example if you want to create castellated holes: http://support.seeedstudio.com/knowledgebase/articles/910767-what-are-plated-half-holes-castellated-holes

yaqwsx, 2020-04-28:
For that case, there is an internal parameter `tolerance` - see https://github.com/yaqwsx/KiKit/blob/4f851a4d1856982ee383ab59ea20f1125a17fb73/kikit/panelize.py#L278-L280

Which enlarges the source area by a given amount. However, it is not available in `extractboard` CLI as I thought the user can make the source area larger manually and it would be less confusing compared to specifying the tolerance.

benjaminaigner, 2020-04-28:
THX for the info, I think I will have to start using the Py Interface :-).

yaqwsx, 2020-04-28:
FIY: I added the requested change in d5f1e0ee1b9f7c3519fe08d1b64988fdf0655832

Also, if you find new commonly used patterns for panelization/automation feel free to properly implement them in CLI a submit a PR.

benjaminaigner, 2020-04-28:
I will do that! THX for that cool tool!

Greetings

closed 2020-04-28 by benjaminaigner