yaqwsx/KiKit #23: missing packages on panelized board

grevaillot, 2020-04-28:
Hi !

I just noticed i've got some missing packages on my panelized board using pip package, and on current master.

[image]
[image]

Is there a way to get some error / verbose traces ? Are you interested by the pcb file ?

Thanks for the awesome tool !

yaqwsx, 2020-04-28:
TLDR: Just specify the source area of your boards such, that it includes all the components.

Longer explanation:

The source area is specified by a rectangle. Every board item, that completely fits inside the rectangle is used for the panel, every else item is ignored. When no source area is specified, it is determined by the board outline automatically. 

Internally, there is a mechanism of tolerance - you can specify a distance by which the source area is enlarged. By default, it is set to 5mm. Currently, there is no way to specify a tolerance from CLI, but I guess it could be handy, so I will implement it.

In the meantime, just specify the source area of your boards such, that it includes all the components.

grevaillot, 2020-04-28:
ah yes ! my bad .. it's working perfectly when properly defining the source area ... awesome :)

yaqwsx, 2020-04-29:
Tolerance is not available in CLI and also documented. Added in 01f966e.

closed 2020-04-29 by yaqwsx