yaqwsx/KiKit #24: Option to change annotations on panelized boards

Kedarius, 2020-04-28:
Would it be possible to add an option to change the annotations on the panelized boards? Similar to what KiKit is doing with net names already. For example: D1->Board_1-D1.  
I would like to create CPL file for the panelized boards so I can get boards made and assembled at one go at JLCPCB...

yaqwsx, 2020-04-28:
It is easy to implement. What would be an ideal user interface for this feature?
I am thinking something like: `--renameref <format>`, where the `format` is a user-supplied string containing `{seq}` and `{orig}`, e.g. `Board_{seq}-{orig}` in your case.

However, this will also change the silkscreen (this is how KiCAD works). And this brings me to a question - once you have a different silkscreen, is it usable, and doesn't it violate JLC-PCB free panelization policy? I would rather write to the support and asked if there is a different way to do it (without renaming the refs).

Kedarius, 2020-04-28:
`--renameref <format>`  as you propose would be great. 

I have no idea if it would violate the panelization policy. All I did find is that:

> 8. What are your requirements for the panelized board?
> Please panelize your boards with stamp holes, V-cut panels are not supported currently.

The idea is that I have very small boards (10mm x7mm) literally with one RGB led, one capacitor and a connector and I would like to have them made and assembled in quantity larger than 50:

> 7. I want to order more than pieces, what can I do?
> You can panelize your boards to realize a quantity of more than pieces, please note there is a panel size limit of not bigger than 480x320mm.

I do not mind paying something extra, I do not want to exploit their cheap offer, I just want the boards to be made and assembled....  I've never ordered a panelized board from them and I have a mixed experience with their support because of their English...
In the worst case I can hide the REF silkscreen and add my own fake REF :-)

yaqwsx, 2020-04-29:
I implemented the feature in 4fd0a270645435bc9f38d93de8e440d1665874bc

Please try it. To install the dev version:
```
pip3 install --force --no-deps git+https://github.com/yaqwsx/KiKit@master
```

Kedarius, 2020-04-29:
Thanks, it is working as expected.

closed 2020-04-29 by Kedarius

bemmbix, 2024-02-13:
Is it possible to set the start variable for the Panelnumber 'n'?

When I tried the panel renumbering with `Board_{n}-{orig}` it always starts with n=0.
I tested `Board_{n+1}-{orig}` but it doesn't work.