yaqwsx/KiKit #27: An error occurred: 'PCB_TARGET' object has no attribute 'GetStart'

patricktlo, 2020-05-01:
Hi,

I compiled the latest version from the repository and I'm getting this error when trying to panelize a PCB.

The example over at /doc/ works, so I'm guessing it's something to do with my design?

I've tried setting an origin and it didn't work.
[test.zip](https://github.com/yaqwsx/KiKit/files/4564276/test.zip)

Thanks!

yaqwsx, 2020-05-01:
Your design uses PCB targets and KiCAD includes them on the Edge.Cuts layer. Therefore KiKit tries to use them to detect board shape but fails. I fixed it in cc56690, so install from master again and it should work:

[image]

One note: your moisture pad has a designator placed far away from the board and KiCAD includes the designator into the item shape - therefore, KiKit excluded the component from the panel. If you move the designator inside the board, it works as expected.

patricktlo, 2020-05-02:
That is great, works perfectly. Thanks for your work!

closed 2020-05-02 by patricktlo