yaqwsx/KiKit #35: Specify path of pcbnew

Gnarflord, 2020-05-15:
So this is kinda related to #21 
I'm using two different versions of KiCAD: The latest stable release from my distribution (ie 5.1.6-1) and a custom self-compiled version which is used in my company.
I'm doing this because we quite like some of the newly introduced features but still want to edit and distribute schematics with the stable release for use by other people.
Now when I want to panelize newer boards I'm greeted with this error:
```
An error occurred: KiCad was unable to open this file, as it was created with
a more recent version than the one you are running.
To open it, you'll need  to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 09/07/19

Full error text:
Unexpected "max_error" in input/source
"test-board.kicad_pcb"
line 50, offset 6
No output files produced
```
That's the same error I get when I try opening newer `kicad_pcb`-files with an older version of pcbnew. They are incompatible because of some added options, `max_error` is just one of these.

Now I can kinda circumvent this by just deleting the offending options from the raw file but it would be nicer if I could specify the pcbnew-binary used by KiKit.

I know, that's quite a niche error case, but this tool has been proven to be invaluable and I'd like it to be just a tad more awesome :)

yaqwsx, 2020-05-15:
Which version of KiCAD will be used is determined by the package available in your Python interpreter. 

I have no idea how SWIG in KiCAD works and where are the packages located when you compile your own Python. To choose the correct version you have to specify the Python library path as an environmental variable before KiKit command (if I remember correctly it should be the `PYTHONPATH` variable). The second option is to somehow get the `pcbnew` module into a virtual env. 

Nevertheless, I think this is something that shouldn't be performed by KiKit. However, I am happy to provide scripts to call KiKit with the desired version/to setup virtual env once we find out how to do it.

yaqwsx, 2020-05-29:
@Gnarflord: Did you succeed with specifying Python path to load specific Pcbnew module? If so would you be willing to share your findings? If so I would like to include the information into the KiKit's documentation.

yaqwsx, 2020-10-28:
Note about running KiKit with a specific version of KiCAD was added to the documentation in 1d131f4.

closed 2020-10-28 by yaqwsx