yaqwsx/KiKit #36: KiKit panel does not inherit design rules.

mayl, 2020-05-16:
Noticed that the panel generated by KiKit does not inherit the design rules from the design being panelized.  This makes it so you cannot run a successful DRC on the panel.  I think it would be nice to have the panel inherit the design rules of the individual board so that you can do a nice, final sanity DRC check on the panel before submitting for fab (would even be cool if KiKit could do that from the command line....). 

Sorry for the flurry of issues/pull requests all at once, obviously I've been using KiKit the past few days and I feel like it's better to track these things as I come across them!

yaqwsx, 2020-05-16:
First of all, I appreciate the feedback. I hope I will manage to resolve all the issues soon.

About DRC rules - currently, I am in the process of finding out if the DRC functionality is available from KiCAD Python API. If not, I guess we are out of luck for the current version of KiCAD and I will try to talk to the developers to add it in the next release.

And the second question - I think that "merging two DRC rules" should inherit the more restrictive rule from both. Are there use-cases where such behavior could be unwanted?

mayl, 2020-05-16:
I'm not necessarily sure I know what the best approach to merging design rules is either.  Taking the most restrictive rules may be a good default option for many cases.

Another option that comes to mind would be to specify a manufacturer specific set of design rules for the panel.  I've read that there is some sort of facility for importing design rules from one PCB to another in KiCAD, but I have not personally tried that.  If this is the case, KiKit could provide template design rules for various manufacturers and the user could specify that as the source of design rules for the panel.  Then the final design rule check "verifies" that the panel is appropriate for the selected manufacturer. This seems in line with your vision for having a per-manufacturer exporters for KiKit as well.

yaqwsx, 2020-07-19:
Small update: There is no DRC-related API in KiCAD 5. There are some massive changes in DRC planned for KiCAD 6 and also DRC for Python API is on version 6 roadmap. Therefore, it makes no sense to build a workaround and we should wait for KiCAD 6.

yaqwsx, 2020-12-04:
@mayl I checked the news in KiCAD nightly. It seems that we are able to implement inheritance of the rules and also (a limited) automated running of the check. Please visit #101, test it, and report any issues you find.

mayl, 2020-12-07:
@yaqwsx,

Thanks for implementing this and appreciate the heads up to give it a try.  Unfortunately I'm away from my PC until at least this weekend so won't be able to give any feedback until then at the earliest. If you're in any rush to get this merged please don't wait on me (obviously)!  I'm excited to give the new features a shot as soon as I can!

yaqwsx, 2020-12-30:
Well, and it is broken again for nightly as the interface we used was removed: https://gitlab.com/kicad/code/kicad/-/issues/6885.

mayl, 2021-01-02:
Rats, thats annoying.  FWIW, I haven't been able to get any of the handful of nightly's I've tried to work here either.

Renha, 2021-03-10:
Could we have DRC rules inheriting and other stuff not supported by the upstream disabled for now (or with an command-line option), for the KiKit to work at least as a simple panelization tool? Currently, I could not use it at all because my projects are following the latest KiCad file formats so I couldn't go with downgrading.

yaqwsx, 2021-11-19:
Hello, I am currently in the middle of adding support for v6 and finally, we can properly transfer design rules from source boards to panels. However, I am currently wondering what should be the way of merging DRC rules for multi-board panels. I would like to hear your opinions! I can imagine we could have multiple strategies, any suggestions are welcomed.

mayl, 2021-11-21:
Hi @yaqwsx  - exciting to hear that v6 has stabilized enough that you are moving towards supporting it.  I unfortunately haven't had any recent projects using KiKit or KiCAD so I don't have any good new merging strategy insights since we last spoke.  I had a couple thoughts I will share in case you find them useful:

In the homogeneous case I think there are two reasonable approaches which both could be useful in different cases:

1) Inherit the design rules from the board being used to make the panel, which is useful to verify the panel doesn't introduce manufacturing tolerences in excess of the design, or
2) Specify a custom or predefined set of design rules for the fabricator (JLCPCB, PCBWay etc.) which is useful to verify that the panel is manufacturerable by the chosen vendor (if for instance the panel is made in CI and you sometimes switch between vendors)  

I think a heterogeneous panel is much more murky, but it seems the options would be to look rule by rule and either 

1) inherit the most difficult to manufacture rule, which verifies that the panel didn't get harder than the hardest board or
2) Inherit the easiest to manufacture rule, which verifies that no single member of the panel drives the overall manufacturabilty more than any other
3) Specify a member board to inherit all design rules from
4) Specify custom or predefined design rules for the panel as in 2 from the homogeneous case

Does KiCAD v6 have the option to scope DRC rules to a specific region of the board?

yaqwsx, 2022-05-09:
With KiCAD 6 custom DRC rules, we can actually easily combine multiple designs with multiple design rules into a single panel and verify that all of the designs follow their individual rules:
- we make the panel's global DRC as weak as possible
- we convert the global DRC rules of individual boards into custom DRC rules operating only on the prefixed names of items
- we transform the individual boards' custom DRC rules, so they operate on the prefixed names of items.

This requires parsing the DRC rules (already implemented) and finding out where the names can appear so we can rename them. However, I am giving this a low priority as I am not confident about the usefulness of this feature.

yaqwsx, 2022-07-03:
There is progress: each board in the design has custom generic design rules that it follows. What remains to add is the support for custom rules.

yaqwsx, 2023-06-27:
Finally, after a year, we implemented the support for custom DRC rules in f3ec40f94101a3315d2543135c162bb75392395e. Therefore, now we can arbitrarily combine designs in a single panel and still yield DRC-passing panel.

closed 2023-06-27 by yaqwsx