yaqwsx/KiKit #37: Docs issue

martiinezz, 2020-05-18:
I have just found this fantastic (at least in theory) project and trying to play with it but I get mixed results.

martiinezz, 2020-05-18:
OK so seems docs are way outdated :(

closed 2020-05-18 by martiinezz

yaqwsx, 2020-05-18:
The documentation you can find in the repository is always relevant for the upstream version of KiKit (the one on GitHub). I release a stable version to Pypi once I manage to test the development version enough. So the documentation is not outdated, your KiKit is. Unfortunately, I haven't set up automatic documentation deployment so you can easily find documentation for the Pypi version. It wasn't high on my priority list as I try to release KiKit often.

Which brings me to the following: `--htabs` and `--vtabs` options were added in the version 0.2.4 released a month ago (the current release is 0.3, released two weeks ago). So you probably have a really outdated version of KiKit. You can find the version of your KiKit installation by calling:

```
pip3 show kikit
```

PS: I added a note about the installation of the upstream/unstable version of KiKit to the README.

martiinezz, 2020-05-18:
Name: KiKit
Version: 0.2.4
Summary: Automatization for KiCAD boards
Home-page: https://github.com/yaqwsx/KiKit

yaqwsx, 2020-05-18:
...so just upgrade it?

```
pip3 install --upgrade kikit
```