yaqwsx/KiKit #39: Issues when --space  is smaller than 2x --radius

martiinezz, 2020-05-19:
For example --space 2 and --radius 1 gives such mousebites:
[image]

But with --space 2 and --radius 0.99 it's fine
[image]

Maybe a warning should be displayed in such case?

yaqwsx, 2020-05-19:
For technical reasons, KiKit does not support single-path milled slots (and for the sake of code simplicity it won't in the near future). I added a check for these parameters in 57b58cf.

martiinezz, 2020-05-19:
In the added code I read 

if space <= 2 * radius:

Shouldn't it be 

if space < 2 * radius:

When space = 2x radius issue is still there

yaqwsx, 2020-05-19:
Ah, my bad... fixed c76c51e

closed 2020-05-19 by martiinezz