yaqwsx/KiKit #40: panel with vertical or horizontal tabs only?

martiinezz, 2020-05-20:
Is there a way to make just vertical or horizontal tabs?

yaqwsx, 2020-05-20:
Just specify `--vtabs 0` or `--htabs 0`

martiinezz, 2020-05-20:
What I mean is panel borders (not tabs) I have this option in EasyEDA that lets me choose left/right - top/bottom or all around border. So left right looks like this:

[image]

yaqwsx, 2020-05-20:
Currently, there is no CLI to generate rails. If you want to, you can write a custom script to do so.

martiinezz, 2020-05-20:
Can you point me where to start? What code should I check? My Python skills are rather rusty :( but can give it a try.

yaqwsx, 2020-05-21:
You can start with an [example script from my other project](https://github.com/RoboticsBrno/RB0002-BatteryPack/blob/master/scripts/panelizeBattery.py). Then checkout panelization [documentation](https://github.com/yaqwsx/KiKit/blob/master/doc/panelization.md).

yaqwsx, 2020-05-28:
As it boils down to different requirements for KiKit, I am closing this in favor of #41 which states the problem more clearly.

closed 2020-05-28 by yaqwsx