yaqwsx/KiKit #43: Missing component and mountingholes in output file

oz1cxp, 2020-06-16:
I have made a pcb 30x50mm, that I panelized with this command:
kikit panelize grid --space 3 --gridsize 2 5 --tabwidth 3 --tabheight 3 --htabs 1 --vtabs 1 --mousebites 0.5 1 0.25 --radius 1 --panelsize 180 120 v4.kicad_pcb panel.kicad_pcb

In the output file one component and all 4 mounting holes in the corners are missing, is there a way i can debug this ?

yaqwsx, 2020-06-16:
Holes in KiCAD have label. Even when the label is hidden, KiCAD counts it into the area occupied by the component. KiKit includes only elements that fully fit into source area - which is by default the board edge if you don't specify the source area. I guess you have labels outside the board edge, hence, the holes are not included.

There are two solutions:
- specify source are explicitly
- specify `--tolerance 20` to your command to increase the source area defined by board edge by 20mm.

oz1cxp, 2020-06-16:
Thank you, that was the problem.

closed 2020-06-16 by oz1cxp

NightSkySK, 2020-08-01:
I would recommend to add this command to example page. I've faced the same issue, until I've find this closed case.

yaqwsx, 2020-08-07:
Added a quite detailed note to the example page.