yaqwsx/KiKit #45: Internal routed shapes disappear in panelized board

WebDust21, 2020-06-25:
So I just discovered KiKit, and after figuring out how to get it installed, I've been quite pleased with how well it handles an otherwise tedious process.  
However, I have an issue.  My pre-panelized PCB (60x20mm) has a few routed slots on the PCB (which I placed on Edge.Cuts).  But after panelization with KiKit, however, these routed slots are missing.  

I'm not _really_ complaining, as it's way easier to re-add the routed slots, than panelize by hand.  Except that copy-paste between instances in KiCad has been broken again, so...

yaqwsx, 2020-06-26:
Please, specify which version of KiKit do you use (stable from PyPi or unstable from Github). Also include the board and the command so I can reproduce the issue.

My bet is that you specify `--radius` larger than your slots half-width. KiKit simulates what is the result of the milling operation and if you have large enough mill, the slot is not milled. The upstream version of KiKit should warn you about this, the stable does not. It's also possible this is has the same cause as #42

WebDust21, 2020-06-26:
It seems that there's something to your suspicion, as I tried a much larger slot, and it appears in the --albeit rounded.
However, my argument is that KiKit should not affect anything already present inside of the PCB outline (slots or otherwise) based on an argument specifying mousebite parameters.

Attached are the input and output files, as well as the commandline.
[KiKit Internal Slots.zip](https://github.com/yaqwsx/KiKit/files/4839228/KiKit.Internal.Slots.zip)

yaqwsx, 2020-06-27:
I can confirm that what I said above is true. I have to add that there was a bug with numerical precision that caused slots with exact width of 2 * radius to randomly change their shape. This was fixed in 5701a008c9cf648afafb91b8b75520c3042c133e. Feel free to try the upstream version. Once I test it properly, I make a release.

If you want KiKit to not touch your slots, just don't specify the radius parameter. The purpose of the radius parameter is to simulate the milling process - e.g. to verify that you don't have too narrow slots that your manufacturer won't be able to produce. If you don't care about this check, simply commit the parameter - in my opinion there is no point of having fillets only on the tabs.

yaqwsx, 2020-07-19:
I consider this issue as resolved. If the problem is preserved even on the recent version of KiKit, feel free to reopen it.

closed 2020-07-19 by yaqwsx