yaqwsx/KiKit #46: Feature request - Tooling holes and Fiducial marks

quixoticgeek, 2020-06-29:
When creating a panel where there is an tooling area along the edge, it would be really useful to be have the option to add tooling holes and fiducial marks. 

It would be great if there was a sensible default for the tooling holes of 3.2mm 5mm from the corner of the panel, and for the fiducial 5mm from the horizontal edge, and 10mm from the vertical edge, (to put them 5mm to the side of the tooling holes). 

Thanks

yaqwsx, 2020-07-01:
Currently, KiKit can add both tooling holes and fiducial. However, this is only available from the Python API, not CLI.

The support should be simple enough and could be implemented together with #41.

Could you provide me with a drawing of the tooling holes? I am not sure if I understand your request for their positioning. Also, is there a widely accepted standard/requirements that should KiKit follow?

quixoticgeek, 2020-07-01:
[panel-example.zip](https://github.com/yaqwsx/KiKit/files/4859387/panel-example.zip)

There is no official standard for the position of the tooling holes and the fiducials. It varies from manufacturing line to manufacturing line. But there seems to be a general consensus that Fiducials should be at least 4mm from the edge (so not covered by the clamps that grip the panel), that there should be 3 of them per side that has SMT components (to allow for measuring both size and orientation). It seems that there is a defacto standard to put the fiducials 5mm to the side of the tooling hole, but I think this may just be convention rather than any official best practice or ISO standard. I have attached an example panel layout with mounting holes and fiducial marks. 

This design document from a UK manufacturer, https://www.norcott.co.uk/wp-content/uploads/TN004_007.pdf, but other companies have different specification. 

I agree it should be implemented along with #41, it may make sense to offer a sensible default with just --addtooling --addfiducial (say 5mm from the corner on both x and y, with top fiducial 5mm further in on the x, and bottom fiducial a further 5mm on that (as in the example attached)), but allow it to be changed it as an command line option --toolinghole 20 6.5 --fiducialtop 25 6.5 --bottomfiducial 30 6.5. Which would match the Norcott document for example.

closed 2020-08-08 by yaqwsx