yaqwsx/KiKit #48: Radius error

G-Pereira, 2020-07-05:
I am having trouble using this tool with a specific board and I don't understand the error it is throwing:
```
An error occurred: Cannot make smaller radius than 0.01 mm
No output files produced
```

**Expected behavior**:
PCB panel would be generated

# To Reproduce

PCB layout that triggers this:
https://github.com/Chameleon-Electronics/programmer-adapter/releases/tag/v1.3.0 (PCB attached)

Source board: <include your `.kicad_pcb` file here>
KiKit command:
```
kikit panelize grid --gridsize 2 2 --vcuts programmer-adapter.kicad_pcb panel.kicad_pcb
```

# Screenshots

Not applicable

# Version
0.5.2

G-Pereira, 2020-07-05:
Here it is the PCB file
[programmer-adapter-pcb.zip](https://github.com/yaqwsx/KiKit/files/4875321/programmer-adapter-pcb.zip)

yaqwsx, 2020-07-05:
Oh, this is a bug that got into the latest release. I fixed it in 3ca046e, to use it, install upstream version. I will make a bugfix release in a few days

G-Pereira, 2020-07-06:
Ok, thank you! :)

closed 2020-07-06 by G-Pereira