yaqwsx/KiKit #51: An error occurred: No intersection found within given distance. No output files produced

Kd5ahl, 2020-07-18:
**Describe the bug**: 
Recieve the error: "An error occurred: No intersection found within given distance. No output files produced"

**Expected behavior**: Panelized board with vcuts

Be Gentle with me please. I'm very new to using this so it's entirely likely I've just messed up the command somehow. 
I realize there would be very few boards on this panel given the size of the panel and the size of the board. 
I'm re-using a command from a previous successful run to try and minimize errors.

I'm trying to get vertical v-cuts on the short ends with some tabs separating the boards, and horizontal v-cuts on the long edges with some tabs separating the boards.

could the issue be that there is a narrower section of the board where the sensor elements are placed?

# To Reproduce

Source board: https://github.com/Kd5ahl/BME-BH1750/blob/master/BME%20Lux%20Sensor-BH1750.kicad_pcb

KiKit command: kikit panelize grid --space 3 --gridsize 3 2 --tabwidth 25 --tabheight 25 --vcuts --radius 1 --panelsize 100 100 BME\ Lux\ Sensor-BH1750.kicad_pcb BME\ Lux\ Sensor-BH1750-Panel.kicad_pcb

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

**> kikit --version
> Usage: kikit [OPTIONS] COMMAND [ARGS]...
> Try 'kikit --help' for help.
> 
> Error: no such option: --version
It's fairly recent though, installed it within the last couple of weeks.**

- source of installation (PyPI or upstream version)

**> PyPI**

- your operating system
**> Ubuntu 20.04**

yaqwsx, 2020-07-19:
You are trying to make vertical tabs, which are wider than your board size (25 mm vs 8 mm). That's why KiKit fails to create the panel. The command you probably want to use is:
```
kikit panelize grid --space 3 --gridsize 3 2 --vtabs 0 --tabheight 3 --vcuts --vcutcurves --radius 1 --panelsize 100 100 BME\ Lux\ Sensor-BH1750.kicad_pcb panel.kicad_pcb
```

Kd5ahl, 2020-07-20:
As is suspected - User Error!

closed 2020-07-20 by Kd5ahl