yaqwsx/KiKit #55: KiKit does not port over the board thickness to the panel

iromero91, 2020-07-20:
Related to issue #36 is that the PCB substrate thickness is not ported over either. That is important for mechanical outputs (STEP/FcSTD) of the board when the board is not 1.6mm thick. Optionally if appending more than one board it could be a good check to add a warning/error if a board is imported with a thickness different to the panel. (If not explicitly set it should take the thickness of the first board appended).

yaqwsx, 2020-07-20:
Currently, there is no support for board thickness in PcbNew Python API. However, extracting the board thickness from the file format itself is pretty straightforward, therefore, I consider to implement this feature before KiCAD 6 release and possibly migrate to proper API afterward.

yaqwsx, 2020-08-05:
There is API for that after all... Implemented in 3fc76d1. There are no changes to CLI as currently only the panelization of a single design is accessible via CLI. Also, there is no option to turn off the check - if you need to mix different board thicknesses to a single panel, you can adjust the thickness of the individual boards in your script.

closed 2020-08-05 by yaqwsx