yaqwsx/KiKit #56: [RFC] Add copper fill

iromero91, 2020-07-20:
Specially for FPC panels it is desirable to fill all the unused areas with copper. The rationale in that case is twofold. in one hand it reduces the amount of etching, on the other it produces a significantly sturdier panel for further processing.

This could be done by simply stretching a lowest priority copper zone over the entire area of the PCB and filling only that zone. I have tried this manually on the finished panel PCB and it has worked. 

One issue though is that PCBNEW does not allow zone priorities lower than 0. There are two ways to work that around. One is to require the user to have all their zones with priority >=1 when using this feature; the other is to automatically bump the priority of all the zones by one (I have no idea how to do this in the Python API)

yaqwsx, 2020-07-20:
This functionality should be easy to implement.

yaqwsx, 2020-08-03:
I implemented the feature in #62. @iromero91, could you test it on your boards?

closed 2020-08-06 by yaqwsx