yaqwsx/KiKit #57: Create a Panel with multiple different design

iharshadec, 2020-07-28:
Hi,
I have been using Gerbertools to create a panel design and recently I found that I am not able to generate 4mm longer tab routing tabs and factory was saying no for the panels. And then I found Kikit and it fixed this problem as now I can make any customize in length that factory supports.

So for single design panel kikit is doing great job. Thank you **yaqwsx** for this. 
Do we have any option to merge multiple different design and have a panel designed for that?

yaqwsx, 2020-07-28:
If you need to put multiple PCBs on a single panel you are expected to write a simple Python script. Specifying the panel shape from CLI would be hell, specifying it from a file would require a custom language - so why not use directly Python, which allows for easy integration with a variety of systems.

See https://github.com/yaqwsx/KiKit/blob/master/doc/panelization.md

yaqwsx, 2020-08-07:
There was no activity in the last 7 days; I consider this as resolved. If you have any further questions/proposals, feel free to contact me or reopen the issue.

closed 2020-08-07 by yaqwsx