yaqwsx/KiKit #59: Panel does not include outside edge vcuts when --space set to 0

samkristoff, 2020-07-28:
**Describe the bug**: 
When running `panelize` command with `--vcuts` and `--space 0` the output panel does not include vcuts on the outside edge of the boards in the panel.  See images in steps to reproduce

**Expected behavior**: 
I expected vcuts to always be added to the outside edge.

# To Reproduce
Run a panelize command like this:
`kikit panelize grid --space 1 --gridsize 2 2 --vcuts --panelsize 100 100 ...`

Notice the vcuts on the outside edge of the board are included (marked with red arrows)
[image]

Now run the same command with --space 0
Run a panelize command like this:
`kikit panelize grid --space 0 --gridsize 2 2 --vcuts --panelsize 100 100 ...`

Notice the outside edge vcuts are missing:
[image]

yaqwsx, 2020-08-03:
I fixed it in dc6a8e2. Please, install the development version and test it.

yaqwsx, 2020-08-13:
There was no activity, the feature was tested. Closing.

closed 2020-08-13 by yaqwsx