yaqwsx/KiKit #64: Version command

iharshadec, 2020-08-13:
Hi,

OS:- Ubuntu 20.04

Can we add command to check kikit version?
I just got notification that v0.6.0 is released so I am trying to update in on my machine but I am not sure if I get latest release installed or not. 
If we can add add -version command then we can always make sure that we are on latest release or not.

yaqwsx, 2020-08-13:
The `--version` was implemented in eca2c64200143e9ef3bc1d8789e64e5a3e8fb910 - part of the v0.6.0 release. Just upgrade your version via `pip3 install --upgrade kikit`.

iharshadec, 2020-08-13:
Yes, 
I got it now. Don't know why but I had to remove kikit and install back it.
Got version command working now.

@yaqwsx *thumbsup*

closed 2020-08-13 by iharshadec