yaqwsx/KiKit #65: Left and right rail V-cut missing

samkristoff, 2020-08-16:
**Describe the bug**: Left and right rail V-cut missing

**Expected behavior**: V-Cut between left/right rails and boards

Panelize a board using 
```
kikit panelize grid --space 0 --gridsize 4 2 --vcuts --railsLr 10 --tooling 5 5 3 $CONTAINER_DIR/$PCB_FILE_IN $OUT_DIR/$PCB_PANEL_FILE; `
```

Source board: N/A
KiKit command:
```
kikit panelize grid --space 0 --gridsize 4 2 --vcuts --railsLr 10 --tooling 5 5 3 $CONTAINER_DIR/$PCB_FILE_IN $OUT_DIR/$PCB_PANEL_FILE; `
```

# Screenshots
If applicable, add screenshots to help explain your problem.
[image]

# Version: 
v0.6.0 via latest docker image

yaqwsx, 2020-08-18:
Thank you for reporting. It is fixed in c6da7fd. I will make a patch release soon.

yaqwsx, 2020-08-18:
Released in v0.6.1

closed 2020-08-18 by yaqwsx