yaqwsx/KiKit #66: Shadowing import in substrate.py

rcludwick, 2020-08-17:
**Describe the bug**: A clear and concise description of what the bug is.

Looking at the following code, the variable substrate shadows the "import substrate" above.

```python

        try:
            substrate = Substrate(edges, bufferOutline)
            self.boardSubstrate.union(substrate)
            self.substrates.append(substrate)
        except substrate.PositionError as e:
            point = undoTransformation(e.point, rotationAngle, originPoint, translation)
            raise substrate.PositionError(filename + ": " + e.origMessage, point)

```

Once ```substrate``` is defined, that causes a namespace conflict with the ```except``` statement as ```substrate``` points to an instance of the class ```Substrate()``` not the package named ```substrate```.

rcludwick, 2020-08-17:
Fix is to change the name of the local var to something that won't shadow something else.   "s" say?

yaqwsx, 2020-08-18:
Fixed in d72e4c0

closed 2020-08-18 by yaqwsx