yaqwsx/KiKit #67: A LinearRing must have at least 3 coordinate tuples

rcludwick, 2020-08-17:
**Describe the bug**: A clear and concise description of what the bug is.
```bash
$ kikit panelize grid --gridsize 2 3  decade.kicad_pcb decade_panel.kicad_pcb

An error occurred: A LinearRing must have at least 3 coordinate tuples
No output files produced
```

**Expected behavior**: A clear and concise description of what you expected to happen.

A panel to have been made using the defaults for a 2x3 grid.

# To Reproduce

Use the board here:  

Create a panel with a 2x3 grid.

https://github.com/rcludwick/decade-box

Source board: <include your `.kicad_pcb` file here>

https://github.com/rcludwick/decade-box

KiKit command:
```
$ kikit panelize grid --gridsize 2 3  decade.kicad_pcb decade_panel.kicad_pcb
```

# Version

0.6.0/master.

Error is hidden behind #66.

yaqwsx, 2020-08-18:
On line 1602 in your `.kicad_pcb` there is a line with a zero length:

```
  (gr_line (start 100.75 73.25) (end 100.75 73.25) (layer Edge.Cuts) (width 0.01) (tstamp 5F017110))
```

This is what makes KiKit yield an error. I wasn't aware that such lines are possible in KiCAD. I added a check for such spurious geometry elements in 41e99b1.

rcludwick, 2020-08-18:
Thanks.  Feel free to add that kicad_pcb to a test suite, if you'd like.

yaqwsx, 2020-08-18:
Released in v0.6.1

closed 2020-08-18 by yaqwsx

rcludwick, 2020-08-18:
Thank you, boards are off to the fab.