yaqwsx/KiKit #71: Not complying with PEP 427 - resources shouldn't be in the module directory

FFY00, 2020-08-27:
PEP 427 says that any file that is not normally installed to site-packages, which is the case of `resources` directory, should be installed to `distribution-version.data` ([here](https://www.python.org/dev/peps/pep-0427/#the-data-directory)).

Solution: python-packaging.readthedocs.io/en/latest/non-code-files.html

yaqwsx, 2020-08-31:
Could you be more specific, what should be changed (a PR would be welcomed)?

I was following the page you refer:

> Files which are to be used by your installed library (e.g. data files to support a particular computation method) should usually be placed inside of the Python module directory itself.

Therefore, I placed the files used by KiKit inside the `kikit` directory. So now I am a little bit confused.

FFY00, 2020-08-31:
Ah, sorry. I didn't read that part, my bad. I just searched for the setuptools documentation on how to the .data directory. The PEP is a bit ambiguous, it does not specify what is usually installed to site-packages. It seems like PyPA says that these files are valid, so let's go with that :blush:

closed 2020-08-31 by FFY00