yaqwsx/KiKit #74: Some footprints are dropped from design when panelized.

samkristoff, 2020-09-07:
**Describe the bug**: Some footprints are dropped from design when panelized.

**Expected behavior**: All footprints from original design carry through to panelized design.

# To Reproduce

Source board: I can share a design that reproduces the issue privately.
KiKit command:
```
kikit panelize grid --gridsize 2 4 --vcuts --panelsize 180 70 $CONTAINER_DIR/$PCB_FILE_IN $OUT_DIR/$PCB_PANEL_FILE; `
```

# Screenshots
Original design has many components, most of which carry over correctly to the panelized design, however the components marked with red arrows are not present in panelized design.
[image]
[image]

Panelized output copper fill indicates componets were there, but the full footprint (with pads and drills) is not there:
[image]
[image]

The missing footprints were the last few added, however I tried adding another copy of the same footprint in pcb_new and it carried through to the panel.

[image]

Moving the footprints from the design sometimes causes them to carry through to the panel.  In the image below I moved U2 and the 2 pin connector.  U2 carried through to the panel but the 2 pin connector did not.  I haven't been able to find a patter to the effect of the position.
[image]

This is a pretty basic design.  Most of the symbols and footprints are custom (clones of stock footprints with altered silk layers).  I wouldn't be surprised if this is something that is only affecting me or this project so I'm happy to do some debugging / troubleshooting, but I'm not sure what to try next so I'd appreciate any suggestions.

Thanks!

# Version
0.6.1 docker image on Windows 10 host.

yaqwsx, 2020-09-07:
Read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md). I guess this is your also your problem - components labels outside the board edge.

This behavior will become more friendly in the near future once #73 will be resolved.

samkristoff, 2020-09-07:
Thanks for the quick response.  That was the issue.  I was able to fix it using the `--tolerance` parameter.

Thank you!

closed 2020-09-07 by samkristoff