yaqwsx/KiKit #75: Fit maximum amount of boards

danielkucera, 2020-09-11:
I have a board of size 56x26mm and I'd like to fit maximum number of them to a panel of 100x100mm.
I can make grid 1x3 but there is still place to place one next to it vertically.
Is this possible? Could it be automatically computed and placed?

yaqwsx, 2020-09-12:
This was discussed several times on KiCAD IRC channel on Freenode. There is no automatic way to do in KiCAD and there probably never will be one. The problem of the automatic fitting is called nesting and it would require to implement one of the algorithms for it. This isn't the problematic part, the problematic part is to come up with good rules for placing tab in irregularly placed boards. Since PCBs are dirt cheap nowadays and usually and regular grid patterns have benefits (e.g., for automated assembly, breaking out the tabs, etc) I decided to not implement such a feature. The time investment in the development, testing, and support is not worth it due to limited use-case.

If you require to place as many boards inside the panel as possible, you can always write a simple single-use python script using KiKit as a library to create and arbitrarily shaped and arranged panel. To start, read [the documentation](https://github.com/yaqwsx/KiKit/blob/master/doc/panelization.md) and a [simple panelization script](https://github.com/RoboticsBrno/RB0002-BatteryPack/blob/8151e037b629feaffc22db870c06a6331899847e/scripts/panelizeBattery.py) from one of my projects.

danielkucera, 2020-09-13:
Fair enough. Thanks.

closed 2020-09-13 by danielkucera