yaqwsx/KiKit #78: Dead Link

perigoso, 2020-09-23:
**Describe the bug**: dead link.

# To Reproduce

in [faq](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)

the link mentioned here
```
My mouse bites are inside the board, what should I do about it?

KiKit's mouse bites offset specifies how much should be the mouse bites put inside the board. The recommended value is 0.25 mm (read about it in this blog post)
```

yaqwsx, 2020-09-27:
Ah, Siemens killed all the user blogs - including the post by Tom Hausherr: Design for assembly. It was a really nice post. There are some parts of it in this terrible PDF: http://www.magazines007.com/pdf/PCBDesign-Oct2013.pdf

Unfortunately, I haven't found a suitable alternative for the link yet. If you have any suggestions, please send me a tip.

Until I find a decent replacement, I will leave the dead link in the FAQ. This way we know what we referenced and what we should replace it with.

perigoso, 2020-09-27:
I really liked the post, that's why I came looking for it and found it was dead, I don't know of a suitable alternative for now, I will keep my eyes peeled.

iromero91, 2020-09-29:
Here is the archived URL in the wayback machine: [https://web.archive.org/web/20150415040424/http://blogs.mentor.com/tom-hausherr/blog/tag/mouse-bite/](https://web.archive.org/web/20150415040424/http://blogs.mentor.com/tom-hausherr/blog/tag/mouse-bite/) When linking external things it may be a good idea to check that it's in the WBM archive, and link that instead. That's what wikipedia does to combat link rot.

perigoso, 2020-09-29:
That's some good insight, didn't even consider that!

yaqwsx, 2020-10-18:
The web archive looks like a good solution. Thanks! Replaced in f3dd86a.

closed 2020-10-18 by yaqwsx