yaqwsx/KiKit #81: ModuleNotFoundError: No module named 'pcbnew'

ihouses, 2020-10-04:
Hi!

I am new in Kicad with python and I dont know how to import pcbnew module. I guess many people will have the same problem :?

I see as well that kicad python console is working in Python 2.7 whereas I am working in my system with python 3.7.

The photos of the panels that you show are looking impressive so I am looking forward to be able to panel some of my tiny PCBs too :)

yaqwsx, 2020-10-04:
Please, search the existing issues first before opening a new issue. This was already discussed in #21 (which is on the first page with exactly the same name). 

TLDR: Until KiCAD 6 is released there is nothing I can do about it. As a mid-term workaround, you can run KiKit via a docker container `yaqwsx/kikit` (also mentioned in [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md))

closed 2020-10-04 by yaqwsx

AdityaSantoshShinde, 2023-09-07:
i am a windows 11 user and i am also facing same problem and i explored #21 problem can you provide me the solution plz