yaqwsx/KiKit #83: Feature Request: Different PCB designs in the same panel

maribu, 2020-10-22:
For the DIY community it could be quite useful to combine different PCB designs in a single panel - especially for assembly: Often setup fees are paid one per panel. If one panel contains all components needed, this fees need to be paid only once.

While this feature might be difficult to implement in a generalized way, this might not actually be needed: Just being able to merge two PCBs either of identical width or of identical height would often be sufficient.

Let's say I have PCBs A, B, and C like this:

```
+-------+ +---+ +-------+
|   A   | | B | |   C   |
+-------+ +---+ |       |
                +-------+
```

I could first create a panel of each PCB, e.g. like this:

```
+-------+ +-------+ +-------+
|   A   |-|   A   |-|   A   |
+-------+ +-------+ +-------+

+---+ +---+ +---+ +---+ +---+
| B |-| B |-| B |-| B |-| B |
+---+ +---+ +---+ +---+ +---+

+-------+ +-------+ +-------+
|   C   |-|   C   |-|   C   |
|       |-|       |-|       |
+-------+ +-------+ +-------+
```

Now those three panels have the same width - all done with the current features of KiKit. If in addition I could now join the 3x A panel and the 5x B panel by explicitly stating the x position of each of the break tabs, I would get to:

```
+-------+ +-------+ +-------+
|   A   |-|   A   |-|   A   |
+-------+ +-------+ +-------+
  |    |      |      |    |
+---+ +---+ +---+ +---+ +---+
| B |-| B |-| B |-| B |-| B |
+---+ +---+ +---+ +---+ +---+
```

And join this in the next step with the 3x C panel, I would end up with:

```
+-------+ +-------+ +-------+
|   A   |-|   A   |-|   A   |
+-------+ +-------+ +-------+
  |    |      |      |    |
+---+ +---+ +---+ +---+ +---+
| B |-| B |-| B |-| B |-| B |
+---+ +---+ +---+ +---+ +---+
  |    |      |      |    |
+-------+ +-------+ +-------+
|   C   |-|   C   |-|   C   |
|       |-|       |-|       |
+-------+ +-------+ +-------+
```

In the worst case the individual PCBs cannot be properly joined, one could generate a panel of each PCB type with a frame and join those panels then at their frames - without break tabs / mouse bites this time.

yaqwsx, 2020-10-22:
Hi, please refer to issue #57. This is already supported via writing a simple Python script. 

TLDR; CLI for such a functionality would be hell and painful to use. We could design a custom file format but why to reinvent the wheel and why not directly use Python?

maribu, 2020-10-23:
Cool, thanks!

closed 2020-10-23 by maribu