yaqwsx/KiKit #85: Solve KiCAD 6 compatibility

yaqwsx, 2020-10-25:
There are some breaking changes in KiCAD 6. Be ready for the release.

Currently known problems:

- `DRAWSEGMENT` has been removed

dave2, 2021-02-11:
Settings are no longer in `BOARD` object, but set at a project level (of which it's not clear there's a python API). Throws errors on Kicad nightlies now about it.

yaqwsx, 2021-02-11:
Thank you for noting; I know about this change. I closely watch KiCAD development. I have opened an issue for it over a month ago: https://gitlab.com/kicad/code/kicad/-/issues/6885. Unfortunately, Seth, who is responsible for the API, is recently not really communicative (e.g., see https://gitlab.com/kicad/code/kicad/-/issues/7134). On FOSDEM Wayne mentioned that the API should be ready by the end of March. Therefore, there's nothing we can do other than wait...

Renha, 2021-04-01:
maybe we should update the issue text to reflect the current state? That would make it much more informative. Also, is there any news? The March has ended.

yaqwsx, 2021-04-07:
Well, the developers seem to not respond to any questions regarding the API. Whenever I raise a question in issue, it is ignored; I even tried to contact them via mailing list, but I haven't received a response in over a week: https://lists.launchpad.net/kicad-developers/msg45028.html

There is the option to start a new Python API on our own, but that's quite punk stuff to do and it will even make the installation of KiKit more complicated... Also, it would be time-consuming. That's not something I am willing to do right now and I will probably wait.

iromero91, 2021-06-08:
I got it to mostly work on 6. Two major issues are: 

## Missing/changed APIs for
* Setting board parameters (as said before, it is missing)
* Setting the board aux origin (it may have just moved, havent looked into it)
* Getting the footprint bound box excluding graphics. I think it moved to `GetFpPadsLocalBbox` from `GetFootprintRect`
* Getting the net name for remapping, it moved from `GetNet` to `GetNetName`

## Segfaults with the current actions:
* Flipping a footprint (experienced in the fiducial placing code. I patched it out by removing the bottom side fiducials in my usecase)
* Creating the copper fill (could not trace the exact call that caused the crash because the stack seems to get trashed).

yaqwsx, 2021-06-08:
FYI: This is the solution for the bounding box: https://github.com/yaqwsx/pcbnewTransition/blob/1c0751062b26f0bc5060c3cdd2e437c689ca6812/pcbnewTransition/transition.py#L22-L27

PS: Seth has merged the code for the new API, so he should be adding the missing API soon. Once he starts working on it, we should move KiKit fast to make it compatible with both v5 and v6 so we have enough time to report problems in KICAD.

UnexpectedMaker, 2021-07-17:
Just wanted to ask how v6 compatibility has been going? It's hard to tell from the changes coming into to the nightly builds if any progress on the outstanding issues are being addressed.

Can we get an updated "state of compatibility/issues"?

I'm super keep to start using this, but I am exclusively on v5.99 for any of my KiCAD work.

Thanks :)

yaqwsx, 2021-07-20:
@UnexpectedMaker Unfortunately, nothing has changed since my last post - we still lack all the API @iromero91 mentioned in the post. We also miss the EESchema API required for outputting the assembly data.

The KiCAD upstream makes really slow to no progress regarding API at the moment.

dave2, 2021-07-21:
KICAD roadmap shows the Python support depends on a coherent schematic object, which has been pushed to v7. I don't see any chance of working Python support in v6 TBH. :(

See https://gitlab.com/kicad/code/kicad/-/wikis/KiCad-6.0-Roadmap#python-support

gwideman, 2021-11-10:
I would very much appreciate an update on this issue. Is it still stalled? If still stalled by the unsettled API, has any news been heard about API progress? Thanks.

yaqwsx, 2021-11-11:
I've been working on this in the last few weeks. I want to have support before for v6 before the release. Currently, the progress was stuck at no progress on the KiCAD side, however, this started to move (at least a little):

- https://lists.launchpad.net/kicad-developers/msg45028.html
- https://gitlab.com/kicad/code/kicad/-/issues/6885
- https://forum.kicad.info/t/python-api-status-in-v6/31929/

It seems that thanks to Jon Evans we should be able to implement at least basic support, the rest has to be hacked. I plan to work on it in the upcoming weeks.

UnexpectedMaker, 2021-11-11:
That's AWESOME news!

bradanlane, 2021-11-11:
I just want to chime in to agree that the progress is greatly appreciated!

I am not of any help on the core side but happy to maintain a test environment.

yaqwsx, 2021-11-21:
Hello guys, the compatibility issues should be resolved now, so please, test it and let me know! More info: https://github.com/yaqwsx/KiKit/pull/238

timcooijmans, 2021-12-11:
Hi, First thank you for the great work!

Making tabs that use mouse-bites for cuts is not working for me on the almost-V6 nightly:
```
An error occurred: module 'pcbnew' has no attribute 'PCB_IO'
No output files produced
```

I'm using:

```
    "cuts": {
        "type": "mousebites"
    },
```

yaqwsx, 2021-12-11:
There was a naming change 10 days ago in the nightly. I committed a blind fix about an hour ago - try updating KiKit. Since I don't have have a proper internet connection at the moment, I wasn't able to pull latest nightly and properly test it (that's why its a blind fix).

timcooijmans, 2021-12-11:
Wow you're quick! Issue solved, thanks again!

closed 2021-12-18 by yaqwsx