yaqwsx/KiKit #86: The board loosing components

lanserge, 2020-10-25:
Basically KiKit is using GetBoundingBox to determine the space occupied by modules but this is not correct way.
The BoundingBox is always bigger! Try it with simple capacitor 0805 and you will get size of (4950000 4292857) that is 4.95mm by about 4.29mm but not 2mm by 1.2mm as correct for 0805 footprint.
I have weird non symmetric footprints that result in BoundingBox bigger than whole board and they all removed by KiKit from panellisation.

I had the same issue with my placing scripts and found the only correct way is to use GetFootprintRect function for the modules.

yaqwsx, 2020-10-25:
The bounding box is bigger than the footprint itself since it includes all references and value labels (as described in [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md) - it also contains information about the overcomming the issue you describe.

Thank you for pointing me to the GetFootprintRect function - I switched to using it in 84d65ac. If no problems will be found within a few days, I'll make a release.

PS: For the future, I would really appreciate it if you followed the issue template GitHub provides and not deliberately ignored it.

closed 2020-10-25 by yaqwsx

lanserge, 2020-10-25:
Thank you for the swift response and the fix. And thank you for pointing out to the "issue template" I did not know about it.