yaqwsx/KiKit #87: NPTH holes present on *.Mask layers

lanserge, 2020-11-03:
The footprint of the NPTH hole that added with addNPTHole function result in holes appear on *.Mask layers (used for pads processing).

**Expected behavior**: 
In the kikit/resources/kikit.pretty/NPTH.kicad_mod file in the PAD line instead of "(layers *.Cu *.Mask)" should be without *.Mask that is:
"(pad "" np_thru_hole circle (at 0 0) (size 1 1) (drill 1) (layers *.Cu))"

# To Reproduce
In KiCad for the panel with mouse bytes generate gerber for F.Mask layer and it will have dots in places where holes.

# Screenshots

If applicable, add screenshots to help explain your problem.
[image]

# Version
KiKit version 0.6.1 installed with pip under Linux debian 4.17.17-1

yaqwsx, 2020-11-06:
Changed in b1d6ee4

closed 2020-11-06 by yaqwsx

RomaVis, 2021-05-18:
@lanserge @yaqwsx Hi! Can you post some comment here which explains why is this an expected behavior **not** to plot NPTH on the mask layer? I find it a bit unusual to want NPTH to be covered with mask...

If this was a requirement from a particular PCB house, then maybe it makes sense to provide an argument which would optionally allow to uncover at least mouse bite NPTHs on the solder mask layer?

lanserge, 2021-05-18:
Before PCB production I am making boards myself using a CNC milling method. There are problems with this method like tiny traces of copper left between pads and such PCBs are very difficult to solder (solder flows and bridging everything). Usually people isolate more around tracks but this takes a lot of time. I use *.Mask as knowledge of where the soldering places are located and do more isolation in those areas. This minimises time for milling and makes boards easy to solder  without even a solder mask. I cannot see another method of guessing from gerbers where the soldering places are. But for mouse bites I don't need to spend time isolating unnecessary areas (they are just holes and not for soldering).