yaqwsx/KiKit #89: automatic placement of auxiliary origin and grid origin in the bottom left corner of a panel

czazbi, 2020-11-10:
**Describe the bug**:
Can you add CLI option to automatically place auxiliary origin and grid origin in the bottom left corner of a panel?

**Expected behavior**:
After panelization is done, both origins should be visible in the left bottom corner.

# To Reproduce

Source board: <include your `.kicad_pcb` file here>
KiKit command:
```
<include the KiKit invocation here>
```

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`
- source of installation (PyPI or upstream version)
- your operating system

yaqwsx, 2021-03-23:
This is fixed in #139. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951. I also suggest moving the discussion there.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx