yaqwsx/KiKit #90: Add CLI option for selection of V-CUT layer

czazbi, 2020-11-10:
**Describe the bug**: 
V-Cut lines are added to Cmts.User layer, but this layer is also used by some KiCad footprints (for example mounting holes)
Because of this gerber file with V-Cut contains not only V-CUT but other lines, circles etc.

**Expected behavior**: 
Add CLI option to allow user to select his preferred V-Cut layer.

# To Reproduce

Source board: <include your `.kicad_pcb` file here>
KiKit command:
```
<include the KiKit invocation here>
```

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`
- source of installation (PyPI or upstream version)
- your operating system

yaqwsx, 2021-03-23:
This is fixed in #139. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

I suggest moving the discussion to the PR.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx