yaqwsx/KiKit #91: CLI option to generate 4 tooling holes instead of 3

czazbi, 2020-11-10:
**Describe the bug**: 
By default "--tooling" option places 3 tooling holes in rails, but some assembly houses expect 4 holes.

**Expected behavior**: 
Please add CLI option to place 4 tooling holes instead of 3.

yaqwsx, 2020-11-10:
Hi, could you provide me with links to fab houses that require 4 tooling holes? I would like to read their specification (so I could possibly adjust KiKit further).

czazbi, 2020-11-10:
Hi,
If I remember correctly these two companies send me assembled panels with 4 holes:
https://printor.pl/en/technology/
http://www.mister.net.pl/index.php?go=montazsmd

But they don't have any documents on their web page.

dave2, 2021-02-11:
In my home-based SMD assembly process, I use four tooling holes for stacking the board+stencil. I could do it with three I suppose, but I'd generally assumed four was more reliable and common practice!

yaqwsx, 2021-03-23:
This is fixed in #139. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx