yaqwsx/KiKit #97: traces/pours on mouse bites for testing an entire panel

arikrupnik, 2020-11-26:
Thank you for writing this piece of software, it's quite amazing. It's remarkable how general it is, handling quite sophisticated cases correctly! I wrote my own Python scripts to generate JLC fab files and my own eeschema parser for BOM generation, but I think I'm going to switch to using yours now!

I have a question about panelization, specifically about including copper layers on tabs. My board can run a self test. I would like to run this test while the boards are still in a panel, to save setup time. For this to work, I need to connect VCCs and GNDs on all boards in a panel. I could do that manually after I run `kikit panelize`, but that defeats the purpose of automation! Is there a way to do that now? If not, may I suggest something like *if `--tabsfrom` specifies a copper layer, fill tab with that layer*?

yaqwsx, 2020-11-26:
Currently, KiKit renames all net names to make them unique during panelization. This is to prevent the air wires from one board to another and also allows you to run DRC on the whole panel.

The requested feature could be implemented by passing a custom renamer, which does not rename the nets for VCC and GND. Then it is easy to assign a net to the top and bottom panel. 

Implementation of such a feature is currently not on my roadmap as I have no use for it. Feel free to open a PR for it. If this feature would make your business easier and save your work and thus you are really interested in having it in KiKit, reach me via e-mail and we can discuss the options.

arikrupnik, 2020-11-26:
Thank you, Jan. I sent an email to your `@honzamrazek.cz` address. Let me know if you don't receive it.

closed 2020-11-26 by arikrupnik