yaqwsx/KiKit #98: Ability to rotate 180 degrees every other copy when panelized

dave4445, 2020-11-30:
Is there a way to rotate 180 degrees every other copy of a board when panelizing?

Example:  If I use '--gridsize 2 1' to place 2 copies of a board one above another, but I'd like to rotate the top one 180 degrees so that there is a common edge for vcuts/bytes.

This would be helpful as one edge of my board has components/traces very near the edge, by rotating one but not the other, the same edge on both copies will be back to back.

kikit, version 0.6.1
Linux (debian 10)

yaqwsx, 2020-12-30:
Recently, I was thinking about some improvements in/redesign of KiKit, and such an option could be easily added after them.

However, I cannot give any promises on the timeline of the implementation as currently, KiKit is not my main job and time is becoming a precious resource for me.

Meanwhile, you can write a custom script to specify the panel; see a simple example: https://github.com/RoboticsBrno/RB0002-BatteryPack/blob/baa010a6cda7d175eb96d8e656043b8ac2444515/scripts/panelizeBattery.py

yaqwsx, 2021-01-24:
This feature was implemented in 418106c610c052a58e031ab1f2beda09fcfa4afd.

Check out examples and documentation for the usage.

closed 2021-01-24 by yaqwsx