yaqwsx/KiKit #103: An error occurred: __init__() got an unexpected keyword argument 'capture_output'

agambier, 2020-12-14:
**Describe the bug**: 

When generating a presentation for a board it fails with the error **An error occurred: __init__() got an unexpected keyword argument 'capture_output'**. Tried on my own board but also on the [RB0002-BatteryPack](https://github.com/RoboticsBrno/RB0002-BatteryPack).

**Expected behavior**: 

Generates the same presentation board as the one present in the documentation.

# To Reproduce

Source board: [RB0002-BatteryPack](https://github.com/RoboticsBrno/RB0002-BatteryPack).

Steps to reproduce:
```
> git clone https://github.com/RoboticsBrno/RB0002-BatteryPack && cd RB0002-BatteryPack
> make
```

KiKit command:
```
kikit present boardpage \
	-d README.md \
	--name "Battery Pack" \
	-b "Charger Board" " " build/charger.kicad_pcb  \
	-b "Battery Board" " " build/battery.kicad_pcb  \
	-b "Connector Board" " " build/connector.kicad_pcb  \
	-b "Charger Board Panelized" " " build/charger-panel.kicad_pcb  \
	-b "Battery Board Panelized" " " build/battery-panel.kicad_pcb  \
	-b "Connector Board Panelized" " " build/connector-panel.kicad_pcb  \
	--repository 'https://github.com/RoboticsBrno/RB0002-BatteryPack' \
	build/web
```

# Screenshots

The complete log file can be download from [here](https://www.swisstransfer.com/d/bac1fa8a-ed4d-487a-8ede-948e0f34c9de).

# Version

Specify which version of KiKit do you use:
- `kikit --version` : `kikit, version 0.6.1`
- `python3 --version` : `Python 3.6.9`
- `which kikit` : `/home/agambier/.local/bin/kikit`
- OS: `Linux mint 19.3`

yaqwsx, 2020-12-14:
This issue is caused by Python 3.6. This Python version is over 4 years old and is not supported by KiKit. Please, use Python 3.7. I will add a check for the Python version in `setup.py`.

closed 2020-12-14 by yaqwsx

agambier, 2020-12-14:
> This issue is caused by Python 3.6. This Python version is over 4 years old and is not supported by KiKit. Please, use Python at least 3.7. I will add a check for the Python version in `setup.py`.

My OS is Linux Mint 19.3 based on Ubuntu 18.04 which is a LTS until 2023 and the python package is 3.6.x. Since it's a long term support I don't plan to upgrade to the latest release of LinuxMint for only one tool.

The most important kikit's feature for me is the panelization (not tried yet) and if the presentation feature doesn't work I'll do it by hand with pcbdraw.

> If you have an interest in KiKit supporting Python 3.6, reach out to me via e-mail and we can discuss the options.

Does kikit need special feature from Python 3.7 ?

agambier, 2021-01-27:
> [..]and it would only cost me the time I could spend on my family, friends, other projects, or new features for KiKit.
> [..]Both of these options support KiKit and make it better. I hope you understand my situation and approach.

I understand and also agree.
As I told you, it's no big deal for me.