yaqwsx/KiKit #104: An error occurred: module 'pcbnew' has no attribute 'ZONE_CONTAINERS'

danielkucera, 2020-12-18:
**Describe the bug**: I'm getting an error:
```
An error occurred: module 'pcbnew' has no attribute 'ZONE_CONTAINERS'
No output files produced
```

**Expected behavior**: Generated panel.

# To Reproduce

Source board: https://github.com/danielkucera/rpi4-pcie-bridge
KiKit command:
```
$ bash -x panelize.sh 
+ PYTHONPATH=/usr/lib/kicad-nightly/lib/python3/dist-packages/
+ kikit panelize grid --gridsize 3 1 --space 7 --vcuts --tolerance 14 rpi4pcie.kicad_pcb panel.kicad_pcb
An error occurred: module 'pcbnew' has no attribute 'ZONE_CONTAINERS'
No output files produced
```

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`
```
kikit, version 0.6.1
```
- source of installation PyPI
- your operating system Ubuntu 20.04
- KiCAD from 
```
deb http://ppa.launchpad.net/kicad/kicad-dev-nightly/ubuntu focal main
```
version
```
Application: Pcbnew

Version: 5.99.0-unknown-a8269b638~106~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.4.0-56-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Dec 12 2020 10:17:27
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.3.0
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

yaqwsx, 2020-12-18:
This is already reported by #85 and it was implemented in #101 (not merged yet).

DavidJRobertson, 2020-12-29:
I noticed you merged PR #101, but this doesn't completely fix the issue when trying to panelize. See https://github.com/yaqwsx/KiKit/blob/e2b9e2966ccf7f2a7bc0fd7eff4b75d72dbc3352/kikit/panelize.py#L241

yaqwsx, 2020-12-29:
I am working on it currently, unfortunately, there some major breaking changes in V6; e.g., https://gitlab.com/kicad/code/kicad/-/issues/6885

So adapting to V6 will take some time.

yaqwsx, 2021-08-23:
Closing in favour of #85

closed 2021-08-23 by yaqwsx