yaqwsx/KiKit #108: Windows command (apparently) not correct

michaelruppe, 2021-01-14:
**Describe the bug**: Windows command in the [installation guide](https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md) produces as *invalid reference format* result.
The command is
```docker run -it -w /kikit -v $(pwd):/kikit yaqwsx/kikit /bin/bash```
which I *think* breaks the docker command syntax. The first instance of `/kikit` may be redundant.

I could only get kikit running as expected by using the following command.
```docker run -it -v "$(pwd):/kikit" yaqwsx/kikit /bin/bash```

# To Reproduce

Follow the installation guide on a Windows machine

---

I don't pretend to be a docker expert - this is in fact my first time using it out of necessity. If this is all the result of my own naivety I apologise in advance.
Thanks for the rad tool! Now that I've got a hello world working I'm really excited to automate my panelisation. Appreciate your work 😄

gwideman, 2021-02-03:
@michaelruppe  I tried the original command, and the command you suggested, and both just return syntax errors:

docker: Error response from daemon: create $(pwd): "$(pwd)" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.

So apparently this needs better quoting in some manner.  FWIW, this was using Windows Command Prompt.

gwideman, 2021-02-03:
OK, the main problem with the supposedly Windows command line in the installation guide is that it uses $(pwd), which is a command that doesn't work in either cmd.exe command prompt or PowerShell.

A command that does work is as follows (for Windows cmd.exe): 
`docker run -it -w /kikit -v %cd%:/kikit yaqwsx/kikit /bin/bash`

What does this do?

- Launches Docker, presenting it with the subsequent argument string, in which %cd% has been replaced by Windows with the Windows current directory.
- 'run' means to invoke Docker's run command
- -i is the "interactive" option that opens (and leaves open) a console
- -t opens a terminal window for the console
- the -v option specifies a Windows directory to mount in the container as the path provided in the -w option
-  yaqwsx/kikit   runs this image file in/as a container
- /bin/bash   gives this command to the running container

So this assumes you start, in Windows, in a command window (or PowerShell), and navigate to the directory containing the xxx.kicad_pcb that you wish to panelize.  Then you run the command above. The expected result is that your existing command window will show a new prompt, like:
`root@d7f25e4a8c9b:/kikit#`

... and you can now type in kikit commands, such as 
`kikit --help`

yaqwsx, 2021-07-20:
This should be addressed in 100d0dc664dd7398f285c24747cba6874856c376.

Also, with KiCAD 6 release and the consequent release of KiKit 1.0 no Docker machinery should be needed on Windows nor Mac.

closed 2021-07-20 by yaqwsx