yaqwsx/KiKit #109: Feature request: Add metadata to rail silkscreen (panelize)

michaelruppe, 2021-01-14:
It's common to add panel information to the rails eg.
 - title
 - version
 - panel dimensions
 - assembly / serial / lot number

since this metadata is all up to the user, having some way to bake the data into a makefile or description file is pretty desirable.

A minimum-viable solution might be to include an option `--add-text myString [textWidth] [textHeight]` that centres the text on a rail. This would allow the user to insert a literal string, or reference a string variable in the makefile.

yaqwsx, 2021-01-14:
I was thinking about this; not hard to implement. Once I find time to look at KiKit, I will implement it.

gwideman, 2021-02-03:
I really need this feature too!  I would wish the parameters to be at least the following:
- myString
- textWidth (assuming this actually means character width factor)
- textHeight
- Left and Bottom position (I don't want to be restricted to middle of the rail, or only one piece of text on the rail)
- Orientation (so it can turn 90 degrees for the side rails)
- Layer  (top or bottom silk or copper
- 
And maybe constrain this text to fit on a rail and not accidentally wind up on a board.

Also a way to print a solid (usually white) silkscreen rectangle on the rail, which can be written in with marker, or stamped.

yaqwsx, 2021-02-03:
With so many requirements and options, I think it is best to implement such a feature after CLI rework. Otherwise, there will be a messy, hard-to-use CLI interface.

If you are really in hurry for this feature, you can write a post-processing script that will be called on the panel to add the text.

I would like to implement this as soon as possible, however, although I enjoy maintaining and improving KiKit, it does not pay my bills. So I struggle to find enough time to implement all the useful and exciting features. If you would like help with writing the work-around script, consider hiring me (or hiring me to implement the complete text feature) and thus supporting KiKit.

gwideman, 2021-02-04:
> you can write a post-processing script that will be called on the panel to add the text

Ah, good point, of course we can run other scripts after Kikit has done the hard part.  I'm motivated!

yaqwsx, 2021-03-23:
This is fixed in #139. It allows to place a simple text via CLI or to automatically invoke a custom post-processing script that can basically append any text. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx