yaqwsx/KiKit #114: --tabsfrom option has partial tabs on outside edges of board

WebDust21, 2021-01-23:
When doing a standard "vtabs/htabs" panelized board with KiKit, the mousebites are only between the PCBs, and not along the outside edges of the panelized PCB, as would be expected.
However, if the "--tabsfrom" option is used, partial mousebites are present all around the outside perimeter of the panelized PCB.  These unnecessarily complicate the PCB manufacture process, not to mention add more cleanup work after the boards are produced.  (NOTE: The only KiKit example shown using "--tabsfrom" also uses a full frame, so these mousebites are connected to the outer frame.  I am not using a full frame, so there is nothing for these mousebites to connect to.)

[image]
Here's an example of a board I've panelized using KiKit and "--tabsfrom."  These little "stubs" are around the entire outside perimeter of the board.  (You can see a connecting "mousebite" in the center of the picture.)

WebDust21, 2021-01-23:
I will clarify: the little "stubs" are where I have placed KiCad drawing marks specifying the mousebite locations.  I fully understand the behavior of KiKit in this case, however it's not what I believe was intended for this function--the regular "htabs" and "vtabs" functions do not have mousebite stubs on the outside perimeter of the board.

yaqwsx, 2021-01-24:
Hi, this is indeed an imperfection in KiKit. The `--tabsfrom` option was intended to use with frame or tight grid panels.

Currently, KiKit cannot distinguish which tabs are "outer" and which are "inner". This will probably change, I once introduce the planned board annotations, however, I cannot give you not estimate of when it will be done.

WebDust21, 2021-01-24:
Umm...but the "htabs" "vtabs" functions seem to know what's inner vs outer.  The only reason I'm using "tabsfrom" is because there is no way to offset the "htabs" "vtabs" manually.  Sometimes I need 2 tabs on the sides, but the upper side tab can't be at the top section of a board, for example (a hollow-center board would snap before the tabs).  Another instance, with an odd shaped board (actually the one pictured above), the (single) horizontal tab needs to be in a specific location to make it even possible to snap the boards apart.

I guess all that to say...I wouldn't need to use "tabsfrom" if there was a way to specify the desired location for the htabs/vtabs.  (Obviously, this would only work for single-design panelization projects.)  An idea perhaps?

yaqwsx, 2021-01-24:
`--htabs` and `--vtabs` generate the tabs automatically - thus KiKit precisely knows which tabs are inner and which are outer. However, to decide which of user lines represents inner and outer tabs is not trivial - e.g., what if the user draws the tab under 45°? When should we consider it inner and when outer? It needs a lot of thinking through, this is why KiKit does not supports it (yet).

As a temporary workaround, you can write a custom script to specify the panel; see a simple example: https://github.com/RoboticsBrno/RB0002-BatteryPack/blob/baa010a6cda7d175eb96d8e656043b8ac2444515/scripts/panelizeBattery.py

tonazza, 2021-03-10:
Hello,

I figured out that setting tolerance=5 in the append procedure caused the board edges to be recognized.
I ended up working a little bit on the script and reaching my desired result.

Since I've put a little effort in it, i thought my script may be useful for someone else in the same situation.
I've put everything in this little repo:

[KiKit Script Example](https://github.com/tonazza/kikit_script_example)

Thanks,
Andrea

yaqwsx, 2021-03-12:
@tonazza Oh, I am sorry that I missed your original question. There were so many new discussions and issues that I lost track.

The default tolerance that CLI uses is 5 mm. The reason, why it does not work with tolerance 0 is #99. Unfortunately, the thickness of the Edge.Cuts line is currently taken into account. In your original script, you specified tolerance 2 (nanometers). KiCAD internally uses nanometers for units, thus all KiKit API also uses nanometers. Only the CLI treats it as mm, so the commands are easily human interpretable.

tonazza, 2021-03-12:
@yaqwsx thank you for the clarification.
...now that I started using the python library (see my previous post above), I will probably do so in all the future cases :)
thanks again for the great job you're doing with KiKit!

yaqwsx, 2021-04-13:
So this was addressed in #139. Given the following annotations:

[image]

You can build a panel like this:

[image]

KiKit automatically decides which tabs from the annotation to use.

WebDust21, 2021-04-13:
Whoa, now we're getting somewhere...!  Thanks!

tonazza, 2021-04-14:
thank you @yaqwsx , great job!!!

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx