yaqwsx/KiKit #115: Component rotation handling

yaqwsx, 2021-01-27:
This issue is based on a comment by @clevertrack1 in PR. This is the original message:

I have been using Kikit a bit and found the component rotation a bit of a pain. There is another GitHub project  matthewlai/JLCKicadTools that uses a database to fix the rotations. Can this database be added into kikit? Also is it possible to add a check that  the lcsc part footprint and the  Kicad footprint match?

yaqwsx, 2021-01-27:
There has been a long discussion #34 about the rotation of the components. A few notes from the discussion:
- There is no single standard for the component rotation,
- what matters is the orientation depicted in the component datasheet.
- If you use the default component libraries from KiCAD the orientation might not match. But it might not match per package but per-component (you can have two manufacturers using the same package but with a different orientation in the reel).
- Therefore, KiKit implements per-component annotation to handle the placement correction.
- matthewlai/JLCKicadTools might not always give you the results you expect and especially, it can *break your existing designs* (e.g., you exported your board, manufactured it, everything is Ok; then you update JLCKicadTools, export again and you get badly assembled board).

There are two solutions:
- create a library that contains all the components JLC PCB uses and use it. 
	- pros: it is 100% safe, you don't need an extra catalog to look up the components
	- cons: it is laborious, your run into the same issues if you decided to switch to a different assembly house (and e.g., use replacement components)
	- I drafter an EasyEDA scraper to build such a library; but it is not completed: https://github.com/yaqwsx/EasyEDAFootprintScraper
- implement something like `kikit bake rotation --rules rules.txt board.sch` which would take a list of rules (e.g., rules from JLCKicadTools) and it would bake them into your schematics in the form of KiKit placement correction annotations. If you find some of the corrections are wrong, you can correct them easily. You make this once during the design stage, thus changes in KiKit cannot break your verified design. When you update the board, only unannotated components will be affected. This is actually on my roadmap, however, I struggle to find time to implement this. Also, it requires KiCAD 6 which will provide API for schematics (implementing for KiCAD 5 is doable, but laborious).
	- pros: You can easily get the results you want, IMHO ideal balance between automation and stability of your designs
	- cons: It is not implemented yet.

About "checking footprints": I consider this as rather utopistic. First, there is no ground-truth about the placement in a nice, machine-readable format. The ground truth is stored in a datasheet, often in pictures. Even when we have a good source of ground-truth; I am not sure what precisely should be changed - that your pads match the datasheet drawing? I think in that case it would be actually better to generate the footprints from the ground truth and then skip the validation.

Also; in my experience, you don't have to be afraid of the wrong component orientation a lot; there is a person in JLC PCB who checks your design, they correct the obvious problems automatically and in the case of any doubt, they contact you. Not 100% error-prone, but it works well.

Note that the above is a proposal, however, I cannot give you any time schedule when it will be implemented. I implement KiKit in my free time and every minute I spent on KiKit is a minute I have to sacrifice from my other activities and family. If you find KiKit useful and you would like to help me find a time for implementing user-requested features, consider supporting me: https://github.com/sponsors/yaqwsx

yaqwsx, 2021-02-01:
FYI: Even the JLCKicadTools mentions the problems with the component rotation: https://github.com/matthewlai/JLCKicadTools/issues/28 Thus, I do not plan to incorporate the fully automatic workflow into KiKit. I will probably implement the bake command to setup the initial corrections which you can manually change if you find it is wrong.

jdaandersj, 2021-02-01:
I like the bake command idea but probably will not use it personally.

One thing I was considering makeing a PR for was that corrected rotations seem to be able to exceed 360. I haven't seen it cause a problem but to me in makes sense to modulo 360 the final result?

mairas, 2021-02-21:
I actually already implemented a JLCKicadTools-like footprint corrections command to KiKit but after reading the discussion in this issue I understand the problem better. Not aiming for a perfect but a useful solution, I wonder would it still be useful to support a fab-specific correction file. Then, the correction steps would be as follows:

1. If the component annotation exists, use that
2. If any of the correction patterns match the LCSC part number, use that
3. If any of the correction patterns match the footprint, use that
4. Finally, use a zero correction (0, 0, 0).

The reasoning would be that even if the footprint-based corrections aren't perfect, there still seem to be some conventions on how different footprints are oriented on the reel. If the correction file was part of the project data (as opposed coming pre-installed with KiKit), it would be up to the user to choose which approach to use (if any).

How does this sound to you? If you agree with the general idea, I can modify my implementation and create a PR.

mairas, 2021-02-21:
OK, I now understand the flaw in my proposal: it's specific to JLCPCB and since the unique part numbering is rather specific to JLCPCB, the approach can't easily be adapted to other fabs...

clevertrack1, 2021-05-27:
I have written an awk script to bake in the rotations. Feel free to use it however you want.
[bake_kikit_rotations .txt](https://github.com/yaqwsx/KiKit/files/6551277/bake_kikit_rotations.txt)

clevertrack1, 2021-05-27:
How to use
gawk -f bake_kikit_rotations.awk kicad,sch >kicad_updated.sch
check that it worked then replace the original with the updated file.

clevertrack1, 2021-05-27:
Fix - correction added at endcomp now added as last field in F section
[bake_kikit_rotations2.txt](https://github.com/yaqwsx/KiKit/files/6551761/bake_kikit_rotations2.txt)

MiggyMan, 2021-11-24:
Given that most of these rotation issues occur per-fab and at a component level can this just be done as a custom field in the symbol much as the lcsc field is used to designate part numbers for jlcpcb orders ?

something like jlcRoataion possibly even with an offset adjustment to account for non-centered parts  ?

Seems like that way you could have all of the information stored in the kicad symbol library in a fab specific manner without having to maintain another database of modifications.

yaqwsx, 2021-11-24:
> Given that most of these rotation issues occur per-fab and at a component level can this just be done as a custom field in the symbol much as the lcsc field is used to designate part numbers for jlcpcb orders ?

This is already available; use `JLCPCB_CORRECTION` (see [documentation](https://github.com/yaqwsx/KiKit/blob/master/doc/fabrication/jlcpcb.md)). This issue is, however, more about adding a way to automatically handle inconsistencies between a given fab house and KiCAD **standard** library. You don't want to edit the standard libraries.

PR #126 (2022-05-08): Implement support for footprint correction patterns
In reference to my comments on Issue #115, I decided to provide a PR for review after all.

As discussed in the issue, the approach of correcting part placements based on footprint identifiers is fundamentally wrong: footprint orientation depends on the individual component type and not the footprint, so rotating per footprint may give you wrong results. Nevertheless, there seem to be semi-strong conventions on how different footprints are oriented on the reels. For example, all SOT footprints seem to be 180° rotated, and most *SO* footprints [... cut]
closed 2022-05-08 by yaqwsx