yaqwsx/KiKit #116: Add support for JLC PCB assembly

yaqwsx, 2021-02-01:
JLC PCB started to support THT components assembly. 

- [ ] Identify if anything has changed in the file format
- [ ] No longer ignore THT components from the assembly

jdaandersj, 2021-02-01:
The help articles on the file format have not changed, and I have just tested a board I had already generated and everything shows up in the correct places etc. Looks like format is unchanged. Hope this helps

yaqwsx, 2021-02-01:
Thanks, therefore, we only need to remove the SMD-only component filter from the code. I think there is no use case for having a switch (you can always use `--ignore`).

jdaandersj, 2021-02-01:
I agree, no need for switch. Definitely like the field to ignore components, anything that permits explicit design intent to be defined is good.

jdaandersj, 2021-02-01:
I have just designed a board to use a THT connector and have found that the position needed correction by half of the length in X. This could just be my footprint but I have only experienced rotation errors previously with SMD components. Perhaps more testing is required.

jdaandersj, 2021-02-01:
In case someone reads this issue and wants to use JLCPCB THT assembly before it is solved, this is my workaround:

Edit the footprint for the THT component in your library. Set the footprint properties 'fabrication attributes' to 'Surface mount'.

Now should work as expected as long as you don't use --forceSMD

yaqwsx, 2021-02-01:
The THT support was implemented in 55484ff. Also, I implemented the ignore flag. Please, test it. I will leave this issue opened until I (or somebody else) order a board with a THT component and verifies there are no problems in the process.

yaqwsx, 2021-02-02:
A note; we should probably skip virtual footprints.

yaqwsx, 2021-03-10:
I received a board with THT components today. Everything seems fine.

closed 2021-03-10 by yaqwsx