yaqwsx/KiKit #118: Support for composite footprints

yaqwsx, 2021-02-02:
PCB modules are popular among hobbyists. These modules often expose 2.54mm pin headers. The hobbyist might want to populate the pin headers. However, here comes the struggle:
- user wants to define the whole module as a single footprint (to preserve spacing)
- assembly house needs to specify such component as multiple different ones (pin headers)

It might be useful to have a KiKit command to "rewrite" such footprints to multiple ones automatically.

blaggacao, 2026-08-27:
My intermediate solution. It's unintuitive as hell:

1. Define a DNP/No BOM/No Position component which only has snap points in the footprint (for distance)
2. Place each header in the schematic
3. Manually ensure all net labels align (even though pin numbers might not)
4. Ignore the fact that schematic pin numbers of my DNP component are not available in the footprint (if they where it might get tricky to avoid DRC rules regarding drill holes at the same location)
5. In the PCB Editor place headers on snap points (relatively easy now). Caveat: Snap Points currently only work when placed on `Cu.[F|B]` layer, even though you _can_ place them on other layers.

[image]

blaggacao, 2026-08-27:
X-Ref: https://gitlab.com/kicad/code/kicad/-/work_items/25215