yaqwsx/KiKit #119: tabsfrom: how do you use this feature? Docs needed.

gwideman, 2021-02-03:
The current docs for tabsfrom appears to be just the one sentence in the Examples page:

"...precise tab placement... Just draw a line in one of KiCAD layers (be careful, the orientation matters - the line has to target the board substrate) and specify option --tabsfrom.  See for yourself, there are some lines already prepared for you in conn.kicad_pcb"

OK, so I looked at conn.kicad_pcb, and I see lines with nearby text labels "TabPosition", but I have no idea what are the salient features of these lines and/or labels that make them recognizable to kikit as specifying tabs.

- Is the text label necessary?
- Is there some significance to the lines appearing on Eco1.User layer versus Eco2.User?
- Do these lines indicate the beginning, middle, or end of the tab?
- "be careful, the orientation matters - the line has to target the board substrate", OK, in what way does the orientation matter?  What does it mean to "target the board substrate"?

yaqwsx, 2021-02-03:
The current implementation is rather hacky, simple, and dumb. I have a plan for redesigning this feature (currently it is on the road map after CLI rewrite). Basically, KiKit would distribute a library of virtual footprints that would serve for PCB annotation - e.g., there should be a footprint that marks "here is my custom tab position".

Meanwhile; all tabs in KiKit are created by the following procedure:
- you pick a point that is the tab origin (the point is supposed to be outside the board)
- you pick a vector marking in which direction the tab should be spanned.
- then KiKit "shoots a ray" from the origin point and tests if it penetrates the board outline. If so, the tab and its cuts are computed. If no penetration occurs within a specified distance, an error is reported.

The lines you draw specify the origin (that is the starting point of the line), the direction of the tab (line orientation), and the maximum ray length (line length). Thus, you should draw your line outside the board perimeter and it should end inside the board. Only the line segments are considered, anything else on the layer is ignored.

gwideman, 2021-02-03:
Wow, thanks for your speedy reply, and I realize that the feature is only a stopgap measure. But...
 
>     * you pick a point that is the tab origin (the point is supposed to be outside the board)

What do you mean by the "tab origin". Does this indicate the corner of a tab? Or is it a point on one edge of the tab? Or?
 
>     * you pick a vector marking in which direction the tab should be spanned.

What do you mean by "span". Surely the tab has to span from one board to the adjacent one, as governed by the --space parameter?
> 
>     * then KiKit "shoots a ray" from the origin point and tests if it penetrates the board outline. If so, the tab and its cuts are computed. If no penetration occurs within a specified distance, an error is reported.

OK, I'm not sure what is the salience of the penetrator ray!

> The lines you draw specify the origin (that is the starting point of the line)
... the starting point of what line?

> , the direction of the tab (line orientation), 

.. how does a line that's perpendicular to the board outline (as in the example) specify the "direction" of the tab?

> and the maximum ray length (line length).
What is this ray (line length). Is it the length of the line drawn on Eco1.User or whatever layer? And if so, what does that specify?

I *think* the main thing I'm missing is that I don't know what specifies where the tab is located relative to the single line we add to the PCB.  If I place a horizontal line crossing the vertical edge of the board, will kikit place the tab centered on that line, or above the line, or below that line?

yaqwsx, 2021-02-03:
Okay, let me try it once again; now with pictures.

When KiKit generates a tab, it generates it based on tab origin and direction. When a tab is successfully generated, it consists out of two components - a piece of a substrate (which will be later appended to the panel) and a cut-line.

So assume we have the following PCB outline (the PCB is below the line, there is a free space above the line):

[image]

Then you specify your tab origin and its direction: 

[image]

This is your input. Now KiKit does its job; it shoots two rays `tabWidth` apart and looks for the intersection with existing substrates. Note that if the ray starts within the PCB, no intersection will be found.

[image]

Once we have the intersections, we can easily generate the tab substrate and the cut:

[image]

This is the basic algorithm for generating tabs. Well, we might also call them "half tabs". KiKit usually generates the half tabs around the board bounding box and then expects that two half tabs in the middle of the panel will merge into a single one. Also, KiKit first generates all the tabs and then merges them in one step to the board substrate. The cut is just a polyline which is in later steps either rendered as a V-cut or via mousebites.

gwideman, 2021-02-04:
Ah-HAH! Well I never would have guessed all that, and it looks pretty clever.  Thanks for detailing all that.

One clarification: "Kikit [...] expects that two half tabs in the middle of the panel will merge into a single one."  

So does that mean that when placing the lines in PCBNew, you have to ensure that on the sides that will face other boards, the lines must be at the same position on both sides?  That is, if boards will be joined by tabs on their vertical sides, then we have to place lines on the left and right side of the board at the same height, so the half-tabs will match?

... which I think means that the example board would not work for that purpose, right?
[image]

yaqwsx, 2021-02-04:
Your assumption is correct. This procedure works best with a tight grid or the planned backbone (now I start to see your use case for tight grid).

gwideman, 2021-02-04:
Thanks for the response.

Is "backbone" an industry term for this feature? Because they seem more like "ribs". :-)

gwideman, 2021-03-16:
An additional clarification:

The tabsfrom command takes two arguments:

--tabsfrom <layername-string> <tablength-float>

The second argument, which I've called tablength, is instead of the --tabwidth and --tabheight arguments, when using tabsfrom.

(The tablength argument was not documented elsewhere that I could find.)

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx